basler/pypylon #86: Error Cannot convert image

JonnySme, 2019-02-18:
Couldn't help with my problem. I wrote a script to get a video stream, it works fine on one computer. On another computer transferred, and after a while I started the program to close with an error. The error can come out immediately, after a minute or after 30 minutes, always in different ways. Jumbo Frames also set to 8192. But I can not find the cause of the error, the error itself:
"Cannot convert image. The passed source image is invalid"

Thanks!

pault587, 2019-02-18:
I had something similar on a USB camera when it needed USB3 but I had a USB2 hub. Maybe your frames are incomplete? Are you USB or GigE?

thiesmoeller, 2019-02-18:
The conversion error has typically a grab error as root cause.

So always check `grabResult.GrabSucceeded()` and log/print `grabResult.ErrorCode` and `grabResult.ErrorDescription`

... and specifically for your interface ( GigE? ) you should read the overall tuning and performance hints  from the camera manual ( including the statistics, that you can read out from the `camera.StreamGrabber` nodemap )

JonnySme, 2019-02-19:
@pault587 @thiesmoeller Thanks for your answers! I use GigEю
What do you mean by "grabResult.GrabSucceeded ()", add a check?

By the way, I'm here in the program PylonV. I set the values for Packet size and Inter-Packet Delay, but for some reason after turning the camera off and on, these parameters are reset. I could not find the reason, can I specify these parameters in the code? (C language)

thiesmoeller, 2019-02-19:
@JonnySme : pleease check the examples in the samples folder or even the main Readme.md.

After retrieving a GrabResult you have to call the method `GrabSucceeded()` method on the result.
It should be True, otherwise it will create followup errors if you push this GrabResult into e.g. the ImageFormatConverter.

Regarding PacketSize and InterPacketDelay: 
All Parameters you modify in the camera are volatile, unless you store them in a [UserSet](https://docs.baslerweb.com/#t=en%2Fuser_sets.htm&rhsearch=userset&rhhlterm=userset). 

And of course you can set all parameters programmatically. For IPD and PacketSize, only extra requirement is setting them before "StartGrabbing()"... as they modify the stream configuration

JonnySme, 2019-02-19:
@thiesmoeller, How to set these parameters programmatically? in the manual I could not find their program code

Thanks!

thiesmoeller, 2019-02-19:
... scroll down ... in the referenced manual page ...

JonnySme, 2019-02-19:
@thiesmoeller 
I beg your pardon, my fault, did not quite understand. But already figured out and programmatically added values for IPD and PacketSize. Added also check GrabSuccess. The error is the same ("Cannot convert image. The passed source image is invalid"), the problem remains, there are already no options how to fix it.

I really do not know how to try to correct this problem, the error is not even fixed, it can immediately go out, and maybe in time

JonnySme, 2019-02-19:
@pault587 @thiesmoeller 

in the terminal (I'm working on Ubuntu) in the process of capturing frames, I now began to receive the following message

Error: 3774873620 The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size.

thiesmoeller, 2019-02-19:
If you install the full Pylon5 for Linux SDK from [baslerweb.com](https://www.baslerweb.com/en/products/software/basler-pylon-camera-software-suite/) you get access to the [bandwidth manager tool](https://docs.baslerweb.com/#t=en%2Fbandwidth_manager.htm&rhsearch=bandwidth%20manager&rhhlterm=bandwidth%20manager), and in the readme hints on tuning a linux system for gige. Together with the camera manual, you can optimze camera and system to really get peak performance.

pault587, 2019-02-20:
https://github.com/basler/pypylon/issues/66

If you haven't solved this take a look at this issue.  Maybe you have slower network on the new computer.

lcflorindo, 2019-05-16:
> @thiesmoeller
> I beg your pardon, my fault, did not quite understand. But already figured out and programmatically added values for IPD and PacketSize. Added also check GrabSuccess. The error is the [...]

Hi! Could you share the solution to set the IPD and PS by code? Thanks in advance.

thiesmoeller, 2025-12-15:
Closing as answered. The "Cannot convert image" error is typically caused by grab failures. Always check `grabResult.GrabSucceeded()` and log error codes. The root cause is usually network/hardware performance issues (buffer underruns) which require network optimization, not a pypylon bug. See camera manual and bandwidth manager tool for tuning guidance.

closed 2025-12-15 by thiesmoeller