basler/pypylon #87: Hi there, Actually i wanna grab an image using software trigger and process it. i have used the code below and i got an error in displaying the image.

Thyagutamizhan, 2019-02-20:
from pypylon import pylon
import time
import cv2
from pyzbar import pyzbar

# Create an instant camera object for the camera device found first.
camera=pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
# Register the standard configuration event handler for enabling software triggering.
# The software trigger configuration handler replaces the default configuration
# as all currently registered configuration handlers are removed by setting the registration mode to RegistrationMode_ReplaceAll.
camera.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(), pylon.RegistrationMode_ReplaceAll,
                                 pylon.Cleanup_Delete)

# Start the grabbing using the grab loop thread, by setting the grabLoopType parameter
# to GrabLoop_ProvidedByInstantCamera. The grab results are delivered to the image event handlers.
# The GrabStrategy_OneByOne default grab strategy is used.
camera.StartGrabbing(pylon.GrabStrategy_OneByOne)
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

# Wait for user input to trigger the camera or exit the program.
# The grabbing is stopped, the device is closed and destroyed automatically when the camera object goes out of scope.
while camera.IsGrabbing():
        # Execute the software trigger. Wait up to 1000 ms for the camera to be ready for trigger.
        if camera.WaitForFrameTriggerReady(1000, pylon.TimeoutHandling_ThrowException):
            camera.ExecuteSoftwareTrigger()

        # Retrieve grab results and notify the camera event and image event handlers.
        grabResult = camera.RetrieveResult(5000)
        if grabResult.GrabSucceeded():
           # count = count + 1
           # print('No. of triggers:', count)

            # Access the image data
            image = converter.Convert(grabResult)
            img = image.GetArray()

            cv2.namedWindow('title', cv2.WINDOW_NORMAL)
            cv2.imshow('title', img)
            '''img1 = cv2.resize(img, (0, 0), fx=0.5, fy=0.5)

            barcodes = pyzbar.decode(img1)
            print(barcodes)
            for bar in barcodes:

                x, y, w, h = bar.rect
                print(x, y, w, h)
                cv2.rectangle(img1, (x, y), (x + w, y + h), (0, 0, 255), 2)
                # cv2.imshow('Image', image)

                if (bar.data.decode('utf-8') != None):
                    print(bar.data.decode('utf-8'))
                bardata = bar.data.decode('utf-8')
                bartype = bar.type
                c1 = c1+1
                print('INFO Found in Barcode:', bardata, bartype)
                # draw the barcode data and barcode type on the image
                text = "{} ({})".format(bardata, bartype)
                cv2.putText(img1, text, (x, y - 10),
                            cv2.FONT_HERSHEY_SIMPLEX, 0.5, (0, 0, 255), 2)
            cv2.imshow('title', img1)
            print('No. of passed:', c1)
            c2 = count - c1
            print('no. of failed', c2)
            k = cv2.waitKey(1)
            if k == 27:
                break
        grabResult.Release()

# Releasing the resource'''
camera.StopGrabbing()

gettting error like this:
Process finished with exit code -805306369 (0xCFFFFFFF)

thiesmoeller, 2019-03-31:
Hi @Thyagutamizhan,

I would recomend to use a python debugger and check the different actions in your program.
( e.g. pycharm )

If you use a pattern like you posted
```
while camera.IsGrabbing():
    if camera.WaitForFrameTriggerReady(1000, pylon.TimeoutHandling_ThrowException):
        camera.ExecuteSoftwareTrigger()
  
    grabResult = camera.RetrieveResult(5000)
    if grabResult.GrabSucceeded():
      ...
```

it is much more efficient to use the image callbacks [ shown in the sample codes ] because here you wait for an image, process it and than trigger the next. You can overlap both activities.

sinasartip, 2020-11-10:
Doesn't StartGrabbing just keep on taking images? I assumed that When You call the Start Grabbing there is no point in then also triggering. 
I am not sure if this is the case though

thiesmoeller, 2020-11-11:
You can use SW triggering if you want SW synchronized image acquisition.
If the camera receives the trigger command it will acquire, process and transmit one frame.

StartGrabbing is only starting the grab engine on your host and switches the overall acquisition engine in the camera on. It then _continously_ waits for new image data.

closed 2025-05-22 by thiesmoeller