basler/pypylon #88: Issue when working with camera.StartGrabbingMax(Count)

Thyagutamizhan, 2019-02-20:
Code:
camera=pylon.TlFactory.GetInstance().CreateFirstDevice()
    camera.StartGrabbingMax(1)
    converter = pylon.ImageFormatConverter()

    # converting to opencv bgr format
    converter.OutputPixelFormat = pylon.PixelType_BGR8packed
    converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
    while camera.IsGrabbing():
        grabresult=camera.RetrieveResult(5000,pylon.TimeoutHandling_ThrowException)
        if grabresult.GrabSucceeded():
            image = converter.Convert(grabresult)
            img=image.GetArray()
            cv2.imshow('title',img)
            cv2.waitKey(0)

Error Message:
AttributeError: 'SwigPyObject' object has no attribute 'StartGrabbingMax'

thiesmoeller, 2019-02-25:
@Thyagutamizhan why did you modify the main instantiation of the camera, that was already working for you into 
```
camera=pylon.TlFactory.GetInstance().CreateFirstDevice()
```
?
When quoting code, please use the corresponding markdown features available from the GitHub edit toolbar.
This helps to read an issue a lot

Thyagutamizhan, 2019-02-26:
> Code:
>  camera=pylon.TlFactory.GetInstance().CreateFirstDevice()

> camera.StartGrabbingMax(1)
> converter = pylon.ImageFormatConverter()
>
>
> # converting to opencv bgr format
> converter.OutputPixelFormat = pylon.PixelType_BGR8packed
> converter.OutputBitAlignment = [...]

Thyagutamizhan, 2019-02-27:
got it, thanku @thiesmoeller

closed 2019-02-27 by Thyagutamizhan