basler/pypylon #89: issue working with UserOutputValueAll feature

Thyagutamizhan, 2019-02-28:
Hi, actually i accessed the UserOutputValueAll  feature of basler acA640-750mc camera. I set it on with some conditions it perfectly works. But its continuosly in on condition even if i switch off the camera. how can i resole it?

thiesmoeller, 2019-02-28:
This model doesn't exist.

Thyagutamizhan, 2019-03-01:
Is there any solutions to get rid of the issue . 
model: acA640-750uc

thiesmoeller, 2019-03-01:
What output lines are you using?
GPIO or opto coupled ?

Example circuits, that you should follow to design your output circuit are in [docs io](https://docs.baslerweb.com/#t=en%2Fcircuit_diagrams_(ace).htm)
And please select your model in the doc system. It will select the right detail docu sections.

thiesmoeller, 2020-11-30:
no feedback by user for 1.5 years

closed 2020-11-30 by thiesmoeller