basler/pypylon #91: Take image array of two cameras simultaneously

jhacsonmeza, 2019-03-02:
Based on the following code:

```
import os
os.environ["PYLON_CAMEMU"] = "2"
import pypylon.pylon as pylon
import pypylon.genicam as geni
#import from samples folder
from configurationeventprinter import ConfigurationEventPrinter
from imageeventprinter import ImageEventPrinter
import time

class CameraBasler(object):

    def __init__(self):
        tlFactory = pylon.TlFactory.GetInstance()

        # Get all attached devices and exit application if no device is found.
        devices = tlFactory.EnumerateDevices()
        if len(devices) == 0:
            raise pylon.RUNTIME_EXCEPTION("No camera present.")

        # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
        self.cameras = pylon.InstantCameraArray(min(len(devices), 2))

        # Create and attach all Pylon Devices.
        for i, cam in enumerate(self.cameras):
            cam.Attach(tlFactory.CreateDevice(devices[i]))
            cam.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(), pylon.RegistrationMode_ReplaceAll,
                                      pylon.Cleanup_Delete)
            # For demonstration purposes only, add a sample configuration event handler to print out information
            # about camera use.t
            cam.RegisterConfiguration(ConfigurationEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
            # The image event printer serves as sample image processing.
            # When using the grab loop thread provided by the Instant Camera object, an image event handler processing the grab
            # results must be created and registered.
            cam.RegisterImageEventHandler(ImageEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
        # Start the grabbing using the grab loop thread, by setting the grabLoopType parameter
        # to GrabLoop_ProvidedByInstantCamera. The grab results are delivered to the image event handlers.
        # The GrabStrategy_OneByOne default grab strategy is used.
        self.cameras.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)
        self.cameraL = self.cameras[0]
        self.cameraR = self.cameras[1]

    # The grabbing is stopped, the device is closed and destroyed automatically when the camera object goes out of scope.
    def takeStereo(self):
        if self.cameraL.WaitForFrameTriggerReady(100,
                                                 pylon.TimeoutHandling_ThrowException) and self.cameraR.WaitForFrameTriggerReady(
                100, pylon.TimeoutHandling_ThrowException):
            self.cameraL.ExecuteSoftwareTrigger()
            self.cameraR.ExecuteSoftwareTrigger()

cams=CameraBasler()
for i in range(10):
    cams.takeStereo()
    time.sleep(1)
```

Given in (https://github.com/basler/pypylon/issues/83) I was wondering if there is a way to get the array of the images taken by each camera, I mean, I do not know if it is possible to do something like `im1, im2 = cams.takeStereo()`.

thiesmoeller, 2019-03-02:
What interface are you using and what is 'simultaneously' in units of time?

There are multiple options to realize your task. 

From the coding level it is quite easy to fetch the images in this routine.

Just call `RetrieveResult(timeout_ms)` after executing the SW trigger for both cameras and extract the image.
Take care to call `Release()` on the Grabresult to return the buffers back into the acquisition engine.

jhacsonmeza, 2019-03-03:
I am using two Basler acA1300-200um cameras in a stereo vision system and I need to capture frame by frame the images of each camera simultaneously and synchronously in order to make 3D reconstruction of a dynamic scene, that is why it is important that cameras take frames at the same time. Looking for how to do this I found the above code example but I need to get the array of images to show and process them.

So, can I do something like this?
```           
def takeStereo(self):
    if self.cameraL.WaitForFrameTriggerReady(100,
                                                pylon.TimeoutHandling_ThrowException) and self.cameraR.WaitForFrameTriggerReady(
            100, pylon.TimeoutHandling_ThrowException):
        self.cameraL.ExecuteSoftwareTrigger()
        self.cameraR.ExecuteSoftwareTrigger()

    grabResult1 = self.cameraL.RetrieveResult(5000)
    grabResult2 = self.cameraR.RetrieveResult(5000)

    grabResult1.Release()
    grabResult2.Release()

    return grabResult1.GetArray(), grabResult2.GetArray()
```

Thanks for your answer.

thiesmoeller, 2019-03-05:
.. First copy out with GetArray and then release, otherwise the buffer doesn't belong to you anymore

jhacsonmeza, 2019-03-05:
I've tried doing this:

```
def takeStereo(self):
        if self.cameraL.WaitForFrameTriggerReady(100,
                                                 pylon.TimeoutHandling_ThrowException) and self.cameraR.WaitForFrameTriggerReady(
                100, pylon.TimeoutHandling_ThrowException):
            self.cameraL.ExecuteSoftwareTrigger()
            self.cameraR.ExecuteSoftwareTrigger()
            
        grabResult1 = self.cameraL.RetrieveResult(5000)
        grabResult2 = self.cameraR.RetrieveResult(5000)
        
        r1, r2 = grabResult1.GetArray(), grabResult2.GetArray()
        grabResult1.Release()
        grabResult2.Release()
        
        return r1, r2
```

But I get the following error:

```
RuntimeException: There is already a thread waiting for a result. : RuntimeException thrown (file 'InstantCameraImpl.h', line 1088)
```

I do not know why.

thiesmoeller, 2019-03-08:
... remove 

```
# The image event printer serves as sample image processing.
            # When using the grab loop thread provided by the Instant Camera object, an image event handler processing the grab
            # results must be created and registered.
            cam.RegisterImageEventHandler(ImageEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
```

In #83 a background acquisition was requested. You asked for a foreground acquisition.
This is the "a thread waiting for a result" ...

binglebai, 2020-03-11:
@jhacsonmeza 
I have the same problem 
could you show you code

`_genicam.RuntimeException: There is already a thread waiting for a result. : RuntimeException thrown (file 'instantcameraimpl.h', line 1071)`

jhacsonmeza, 2020-04-04:
Hi @binglebai this snippet of code works for me to acquire images simultaneously from two USB cameras and display them with OpenCV.

```
from pypylon import pylon
import numpy as np
import cv2

cv2.namedWindow('Acquisition', cv2.WINDOW_NORMAL)
cv2.resizeWindow('Acquisition', 1280, 512)

tlFactory = pylon.TlFactory.GetInstance()
devices = tlFactory.EnumerateDevices()
if len(devices) == 0:
    raise pylon.RUNTIME_EXCEPTION("No camera present.")

cameras = pylon.InstantCameraArray(2)

for i, camera in enumerate(cameras):
    camera.Attach(tlFactory.CreateDevice(devices[i]))

# Starts grabbing for all cameras
cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly, 
                      pylon.GrabLoop_ProvidedByUser)

while cameras.IsGrabbing():
    grabResult1 = cameras[0].RetrieveResult(5000, 
                         pylon.TimeoutHandling_ThrowException)
    
    grabResult2 = cameras[1].RetrieveResult(5000, 
                         pylon.TimeoutHandling_ThrowException)
    
    if grabResult1.GrabSucceeded() & grabResult2.GrabSucceeded():
        im1 = grabResult1.GetArray()
        im2 = grabResult2.GetArray()
    
        # If ESC is pressed exit and destroy window
        cv2.imshow('Acquisition', np.hstack([im1,im2]))
        if cv2.waitKey(1) & 0xFF == 27:
            break

cv2.destroyAllWindows()
```

philipseo7924, 2022-05-30:
@jhacsonmeza Hi. i have the same problem. i figured out to change the parameters of StartGrabbing function. when you use software trigger, you must set the second parameter to GrabLoop_ProvidedByUser. then you can get the results from RetrieveResult.
Hope to help you.

aruna-ram, 2022-07-05:
Hi there, @jhacsonmeza are you able to extract synchronous data from two USB cameras using software triggers only? Reading around there seems to be conflicting information on whether this is possible or not. Many places suggest using a hardware trigger e.g. https://github.com/basler/pypylon/issues/9 , https://github.com/basler/pypylon/issues/162 , https://github.com/basler/pypylon/issues/470

jhacsonmeza, 2022-07-05:
Hi @aruna-ram as I mentioned previously, with this code you can simultaneously acquire frames from two USB cameras:

> Hi @binglebai this snippet of code works for me to acquire images simultaneously from two USB cameras and display them with OpenCV.
>
> ```
> from pypylon import pylon
> import numpy as np
> import cv2 [...]

In this example, I used OpenCV to visualize the acquired frames, but this isn't required. The key part in the code is using:
```
cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly, 
                      pylon.GrabLoop_ProvidedByUser)
```
which allows you to control/write your own loop (enabled with the option `pylon.GrabLoop_ProvidedByUser`) to acquire the frames and convert them to numpy arrays with `.GetArray()`.

Note that this code only works considering that you're not using an external trigger signal.

Let me know if it works for you!

jvitordm, 2023-05-02:
@jhacsonmeza 

Thanks for the code above! Is it possible to run it with a SoftwareTrigger? Sould I then get better synced results compared to the ones above?

Thanks!

jhacsonmeza, 2023-05-03:
Hi @jvitordm I think if you want to use SoftwareTrigger you need a different implementation than that I provided. For my specific use case, my version worked better than the SoftwareTrigger.

jvitordm, 2023-05-03:
Hi @jhacsonmeza thanks for the reply! Indeed your code is working quite well for me with 4 cameras, just wanted to push further to check if with a trigger it could get even better... do you happen to have or know such an implementation? I made some tests, but still couldnt make it work as reported here https://github.com/basler/pypylon/issues/605

Thanks!

jhacsonmeza, 2023-05-05:
You're welcome! @jvitordm 
Just to be 100% sure I understood what you want to do. Are you using an external trigger signal? From an Arduino or something like that?

jvitordm, 2023-05-05:
@jhacsonmeza Not at the moment, I wanted first to test a software trigger for multiple cameras (like in https://github.com/basler/pypylon/blob/26009c3f0b4490ab06800da6feebf7e15349b5cb/samples/grabstrategies.py) to see if I can enhance the sync performance. The external trigger would be my last resort.

jhacsonmeza, 2023-05-05:
Oh, I see. In that case I'm sorry I can't help that much. I know how you can use an external trigger signal but I don't know how to use the software trigger. As you did, I just tried the code example you pointed out, and that's it.

closed 2023-05-16 by jhacsonmeza