basler/pypylon #92: How to run a c++ sample code in python

zamarjavier, 2019-03-06:
Hello,

I'm new using basler cameras. I'm trying to take photos with a usb camera using pylon5 and executed from python. 
The idea is to run the c++ code to take a photo (from python), store the photo in a python variable and then use this variable in python to  process it. How can I do it?

I thought in use a tcp conexion between c++ code and python code, but maybe the process to follow is easier.

Can anyone help me?

Thank you so much.

thiesmoeller, 2019-03-08:
Hi @zamarjavier,

this is the python binding to basler cameras. You can do the same things you can do using the C++ APIs.

Look into the samples folder to get an idea, of what is possible.

Or is there a strict reason for you to use C++ _and_ python?

spy14414, 2019-03-28:
You can wrap c++ ​​code into a dynamic link library.Then  you just need to call this dll in python.

thiesmoeller, 2020-11-30:
off topic issue, as this is the python binding.

closed 2020-11-30 by thiesmoeller