basler/pypylon #94: How to capture images with different exposures?

chuong, 2019-03-20:
I would like to capture an HDR image by combining images of different exposures. How such images can be captured efficiently using Pypylon?

Thanks

chuong, 2019-03-20:
This is similar to Sequencer feature available in C/C++ interface:
https://docs.baslerweb.com/#t=en%2Fsequencer_(gige_cameras).htm

How can this be done in Python?

thiesmoeller, 2019-03-20:
[See in this issue](https://github.com/basler/pypylon/issues/67#issuecomment-457585961)

chuong, 2019-03-21:
Great pointer. Thanks @thiesmoeller.

chuong, 2019-03-27:
Hi @thiesmoeller, the python example you mentioned is for USB3 camera and it doesn't work for my GigE Blasler camera.

I tried to translate C++ code from https://docs.baslerweb.com/#t=en%2Fsequencer_(gige_cameras).htm to python but there are missing python functions.

thiesmoeller, 2020-11-30:
@chuong as all features are downloaded from the camera itself, in principle there can't be a missing function

closed 2024-04-21 by SMA2016a