basler/pypylon #97: Access point cloud from python

LaurentBimont, 2019-03-25:
Hi,
I am using a Basler TOF software with Basler tof640-20gm hardware. I would like to access the point cloud via python but I am struggling in doing so. When I do 

`first_device = pylon.TlFactory.GetInstance().CreateFirstDevice()`
`instant_camera = pylon.InstantCamera(first_device)`
`instant_camera.Open()`
`grab_result = instant_camera.GrabOne(1000)`

`image = grab_result.GetArray()`

I only have a gray image. However in the software Basler TOF I can see the 3D image, How can I access it ?

Thank you for your answers.

spy14414, 2019-03-28:
The image you got maybe a intensity image. In my opinion，You can get the point cloud by processing the intensity image. The code in c++ is:
`			// Calculate parameters for projecting 3D data to 2D image
			// using a pinhole camera model.
			Mat A = Mat::eye(3, 3, CV_64F);  // Intrinsic camera parameters
			A.at<double>(0, 0) = m_f;
			A.at<double>(1, 1) = m_f;
			A.at<double>(0, 2) = parts[0].width / 2.0;
			A.at<double>(1, 2) = parts[0].height / 2.0;
			Mat F = Mat::eye(3, 3, CV_64F);  // Used for flipping the image
			F.at<double>(0, 0) = -1;
			F.at<double>(1, 1) = -1;
			F.at<double>(0, 2) = (double)(width - 1);
			F.at<double>(1, 2) = (double)(height - 1);
			A = F*A;
			// m_R and m_t are the extrinsic parameters
			const double cosphix = cos(m_phiX);
			const double sinphix = sin(m_phiX);
			const double cosphiy = cos(m_phiY);
			const double sinphiy = sin(m_phiY);
			double r[3][3] = {
				{ cosphiy, 0, sinphiy },
				{ sinphix*sinphiy, cosphix, -sinphix*cosphiy },
				{ 0, 0, 0 } };
			m_R = Mat(3, 3, CV_64F, r);

			Mat distCoeffs = Mat::zeros(Size(5, 1), CV_32F); // No lens distortion
			// Perform the projection
			std::vector<Point2f> points2D;
			projectPoints(cloud, m_R, Mat(m_t), A, distCoeffs, points2D);

			// We create two images. For the first one, we assign each point its corresponding intensity value.
			// For the second image, we assign each point the distance to the pinhole camera.
			Mat intensityMap = Mat::zeros(height, width, CV_16UC1);
			Mat distanceMap(height, width, CV_64FC1);
			distanceMap.setTo(std::numeric_limits<double>::infinity());
			Mat Range = Mat::zeros(height, width, CV_16UC1);
			//Range.setTo(std::numeric_limits<double>::infinity());
			for (int row = 0; row < height; ++row)
			{
				int idx = row * width;
				for (int col = 0; col < width; idx++, col++)
				{
					double x = cloud.at<float>(idx, 0);
					if (x < 1e6)
					{
						double y = cloud.at<float>(idx, 1);
						double z = cloud.at<float>(idx, 2);

						uint16_t range;
						if (z <= 13320)
						{
							range = (unsigned int)z / 13320 * 65536;
						}
						cout << z << endl;
						//Range.at<uint16_t>(col, row) = range;

						double dx = m_t.x - x;
						double dy = m_t.y - y;
						double dz = m_t.z - z;
						double distance = sqrt(dx*dx + dy*dy + dz*dz);
						int u = (int)(points2D[idx].x + 0.5);
						int v = (int)(points2D[idx].y + 0.5);
						if (u > 0 && u< width && v > 0 && v< height && distanceMap.at<double>(v, u) > distance)
						{
							// (u,v) are the coordinates to which the 3D point is projected, (col, row) are the 
							// coordinates of the pixel in the original sensor image.
							// By comparing the current distance with the distance already stored we ensure that, if multiple 3D points 
							// are projected to the same destination pixel, the nearest point wins.
							intensityMap.at<uint16_t>(v, u) = intensity.at<uint16_t>(row, col);
							distanceMap.at<double>(v, u) = distance;
							//cout << v << endl;
						}
					}
				}
			}

			Mat mask;  // Used for masking points without depth information
			compare(distanceMap, std::numeric_limits<double>::infinity(), mask, CMP_GE);
			distanceMap.setTo(0, mask);
			Mat grey;
			minMaxLoc(distanceMap, NULL, &max);
			distanceMap.convertTo(grey, CV_8UC1, 255.0 / max);
			Mat rainbow;
			applyColorMap(grey, rainbow, COLORMAP_RAINBOW);
			rainbow.setTo(0, mask);
			imshow("Point Cloud (intensity)", intensityMap);
			imshow("Point Cloud (distance)", rainbow);
`
You can get it in the (The way you install tof)Basler\ToF\Samples\OpenCV\ShowPointCloud .

thiesmoeller, 2020-12-02:
main question has been answered

closed 2020-12-02 by thiesmoeller