basler/pypylon #99: Camera external trigger

ruggiamp, 2019-03-25:
Hello,
I would like to capture images with a Basler camera (Basler ACE ACa2040-90 um NIR) using an external hardware trigger. 
Does anyone have any sample code (in Python) that could help me understand how to select the acquisition mode, how to tell the camera to wait for 'frame burst' and 'frame start' triggers, how to set the trigger source, trigger delay, exposure time and so on?
I looked for it into pypylon documentation, c++ API, ..., but I really don't have a clue.
Any suggestion would be more than welcome! :)

Thank you in advance
Ruggero

nikkadim, 2019-03-27:
camera.TriggerMode.SetValue("On")

thiesmoeller, 2019-03-31:
```
camera.TriggerSelector.SetValue("FrameBurstStart")
camera.TriggerSource.SetValue("Line1")
camera.TriggerMode.SetValue("On")

```

ruggiamp, 2019-04-02:
Many thanks for your answer!

I managed to grab a picture!
Still, I am wondering how to have different cameras acquiring pictures at the same time with external triggers . As I understand 'RetrieveResult' function only grabs one result at a time but I would like something that can cope with different cameras triggered independently, even at the same time...

The Basler Manual is good, but doesn't say anything about functions as 'RetrieveResult' or 'RegisterConfiguration' .

nikkadim, 2019-04-02:
@ruggiamp  as you said, Basler Manual is good, I took following from that manual:

if you know camera IP make a DeviceInfo for that camera and make an object for particular
```
tlFactory  = pylon.TlFactory.GetInstance()
device_class = "BaslerGigE"
di = pylon.DeviceInfo()
di.SetDeviceClass(device_class)
di.SetIpAddress("192.168.2.22")

# connecting to selected camera
camera = pylon.InstantCamera(tlFactory.CreateDevice(di))
```

ruggiamp, 2019-04-02:
If I set cameras for hardware external trigger, call the function 'cameras.StartGrabbing()' and apply to all the cameras an external trigger signal at the same time, are the cameras starting grabbing synchronously?
As I understand images are then collected asynchronously by the for cycle with the 'RetrieveResult' method, but I was wondering if they are at least taken at the same time.
Here the part of the code I am using (with two cameras) to get the pictures:

 
    NumberOfPictures = 2
    Imagenum = 0  ### track image number
    print('Start acquiring')
    Images = []
    
    cameras.StartGrabbing()
    print('Waiting for trigger')

    for Imagenum in range(NumberOfPictures):
    
        ###  A timeout of 10000 ms is used.
        grabResult = cameras.RetrieveResult(10000, pylon.TimeoutHandling_ThrowException)
    
        ### Print the index and the model name of the camera.
        cameraContextValue = grabResult.GetCameraContext()
        CameraSerialNumber =  
        cameras[cameraContextValue].GetDeviceInfo().GetPropertyValue('SerialNumber')
        print("Camera acquiring", SerialNumToName [CameraSerialNumber[1]] )
    
        if grabResult.GrabSucceeded():
            print('Picture number ', Imagenum)
            img = grabResult.Array
            print("Max Intensity:", numpy.amax(img))
            Images.append(img) 
        else:
            print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
            grabResult.Release()

    print('Acquisition process completed')
    print('Images meant to be taken: ', NumberOfPictures)
    print('Images succesfully taken: ', len(Images))

thiesmoeller, 2019-04-03:
Pylon's low level grab engines do capture what the cameras transfer.
RetrieveResult will only make this data available to your application.

If you hw-trigger in sync the images get acquired in sync also

ruggiamp, 2019-04-03:
Great, thank you for the answer!

I still can't understand completely how it works at a buffer level. For example, I would like to be able to acquire many pictures with the same cameras first (using multiple hardware triggers) storing them in a buffer or so and then retrieving those pictures and saving them.
If I call  'cameras.StartGrabbing(pylon.GrabStrategy_OneByOne)' and then 'RetriveResult()', I just get one image only (one for each camera), no matter how many hardware triggers I give to each camera. And including 'RetriveResult()' in a for cycle does not help...

What am I missing?

thiesmoeller, 2021-05-28:
Closed due to long time no further action.
Using multicamer is explained in basler/pylon-samples repo

closed 2021-05-28 by thiesmoeller