basler/pypylon #100: camera options in "CInstantCameraArray"

JonnySme, 2019-03-28:
Hello! I use "CInstantCameraArray" to access the camera. For example, the parameter "GevSCPSPacketSize". I try **"cameras.GevSCPSPacketSize.SetValue(1500);"**, get error: **'class Pylon::CInstantCameraArray'** has no member

"CInstantCameraArray" I use it because I have several cameras, and I need to connect to a specific camera.

Thanks

thiesmoeller, 2019-03-31:
Hi @JonnySme ,

are you programming in C++? This is the python bindings issue tracker.

Anyhow. The InstantCameraArray is not capable to access all camera in parallel with a single call.
It has array semantics, so you can write something like:
`cameras[0].GevSCPSPacketSize.SetValue(1500)`
to set the Packetsize for the first camera in the array.

thiesmoeller, 2020-11-30:
no feedback sind 1.5 years

closed 2020-11-30 by thiesmoeller