basler/pypylon #102: Memory Leak

Eyshika, 2019-04-01:
Am using it with pylon in the gui, where 1 click starts the camera video, and then process opencv fuctions on each frame, but it gives memory leak and crashed the system.
The way it is written is : 
`from pypylon import pylon 

def acquire_movie(self,numFrames,checkHsize,checkWsize):
        movie = []
        
        self.checkHsize = int(checkHsize)
        self.checkWsize = int(checkWsize)

        ##for infinite loop (but crashes)
        #self.cap.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

        # for particular number of frames
        self.cap.StartGrabbing(numFrames)
        
        converter = pylon.ImageFormatConverter()

        #converting to opencv
        converter.OutputPixelFormat = pylon.PixelType_BGR8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

        while self.cap.IsGrabbing():
            self.grabResult = self.cap.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
            if self.grabResult.GrabSucceeded():
                
                image = converter.Convert(self.grabResult) #cv.cvtColor(self.grabResult.Array, cv.COLOR_BAYER_BG2RGB)
                self.rgb = image.GetArray()
                
              #  #some opencv process
                movie.append(self.rgb)
        return movie

class StartWindow(QtGui.QMainWindow):

class MovieThread(QtCore.QThread):
    #separate thread to keep imshow running properly
    def __init__(self,camera,checkHsize,checkWsize):
        super().__init__()
     
        self.camera = camera

    def run(self):
        
        self.camera.acquire_movie(3000,self.checkHsize,self.checkWsize)    
    
`

am doing multithreading to avoid memory issues, but it is still doing it. What is the reason ? something wrong in pypylon package ?

thiesmoeller, 2019-04-05:
Hi @Eyshika ,

as a quick look at your code:

a)
why do you make grabResult an instance variable? 
`self.grabResult = ...`

when you use it only in this little scope this is not needed.

b)
you have to release the grabResult activly back to pylon grab engine. Otherwise you run out of buffers after some frames.

```
while self.cap.IsGrabbing():
        grabResult = self.cap.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        if self.grabResult.GrabSucceeded():
            
            image = converter.Convert(self.grabResult) #cv.cvtColor(grabResult.Array, cv.COLOR_BAYER_BG2RGB)
            grabResult.Release()
            self.rgb = image.GetArray()
            
            #  #some opencv process
            movie.append(self.rgb)
```

we found no way to release the buffer automatically when you write the code like above. There is a pattern with the python contextmanager which works...

```
with cam.RetrieveResult(1000) as result:
        img = result.GetArray()
        ....

```

Eyshika, 2019-04-08:
@thiesmoeller  Thank you for your suggestions. Can you please explain what does the value passed to cam.RetrieveResult means ? in the examples i saw its 5000, so I used 5000 but you are using 1000 here, can you please explain what does this value refer too, I know it's not number of frames, since number of frames are being passed to cap.StartGrabbing().

thiesmoeller, 2019-04-08:
It's the time to wait for a result in ms.

closed 2020-11-30 by thiesmoeller