basler/pypylon #105: WaitForFrameTriggerReady throws exception in samples/grabstrategies .py

yves-surrel, 2019-04-05:
Windows 10, acA2500gm, pypylon 1.4.0

When I run samples/grabstrategies .py, I get:

`TimeoutException: Waiting for frame trigger ready timed out. : TimeoutException thrown (file 'instantcameraimpl.h', line 1666) `

Does it come from the camera itself?

thiesmoeller, 2019-04-05:
Hi @yves-surrel, when posting error messages, please also post the full backtrace, so it is easier to from which line in the sample the error is triggerd.

> Does it come from the camera itself?

No. ... or only indirectly. Pylon generates this exception, when the FrameTriggerReady signal is not coming in the given timeout. Here 200ms.

As a quick check:
   You should check if a longer time helps. The samples do not work with all cameras out of the box....

```
# Issue software triggers. For each call, wait up to 200 ms until the camera is ready for triggering the next image.
for i in range(3):
    if camera.WaitForFrameTriggerReady(200, pylon.TimeoutHandling_ThrowException):
        camera.ExecuteSoftwareTrigger()
```

yves-surrel, 2019-04-06:
Hi Thies

Thank you for the answer. I had already tried to switch to 1000 ms (although 200 ms seems already huge for that kind of signal), but it did not change anything.
With a Pilot camera I also have, it works (not an obvious code problem), so I am still stuck with the acA2500.
Any other attempt to try (e.g. one-liner asking for this signal…) ?

Yves

thiesmoeller, 2021-05-28:
Camera could have a very long exposure time or be configured for trigger.
Closed for no further action after long time

closed 2021-05-28 by thiesmoeller