basler/pypylon #108: Change pixel format with using emulator

oak-tree, 2019-05-01:
Hey, 
I'm trying to make my images on emulator colorful. As suggested in Basler docs, I changed the `camera.PixelFormat`.

## What I did:
In order to do so first, I fetched, the available format from emulator using: 
`camera.PixelFormat.Symbolics` and got 
`('Mono8', 'Mono16', 'BGRA8Packed', 'BGR8Packed', 'RGB8Packed', 'RGB16Packed')`
`camera.PixelFormat.SetValue(<the value>)

## The problem
Then, expect  the `Mono` values, When I try to use the `colorful` formats I get 
`Process finished with exit code -1073740791 (0xC0000409)`  as soon as I run the command         `camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)`

It does work the `Mono8` values. 
Any ideas about how to overcome and use colorful images within the emulator?

Thanks,

## extra code?
  ```
camera.ImageFilename = img_dir
  camera.ImageFileMode = "On"

  # disable testpattern [ image file is "real-image"]
  camera.TestImageSelector = "Off"
  camera.Height.SetValue(416);
  camera.Width.SetValue(1248);

  
  camera.PixelFormat.SetValue("BGRA8Packed")
 camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly
```

thiesmoeller, 2019-05-03:
The payloadsize is not updated correctly. Only when Pixelformat is read once ....
Fixed in pylon >= 5.2

But we have a workaround for you:

```
camera.PixelFormat.SetValue("BGRA8Packed")
camera.PixelFormat.GetValue()
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
...

```

oak-tree, 2019-05-05:
@thiesmoeller thanks! the reading the `PixelFormat` once does the trick

closed 2019-05-05 by oak-tree