basler/pypylon #109: Decimation control

niallodowd, 2019-05-08:
Hello, I have a Basler Ace acA4600. The normal framerate is 7.4fps and I can decimate the input in the AOI commands in pylon to 2x2, which returns a framerate of ~27 fps. I'd like to see this functionality controlable in python, so I can change the decimation automatically, and not have to use pylon.

Is this functionality incorporated in pypylon? If not, please add!

Thanks

thiesmoeller, 2019-05-09:
All features of any Basler device that you can control  in pylon viewer are controllable from pypylon.

 syntax is always like 

`camera.Gain = 42`

If the feature is e.g. Gain

Take a look in the samples of how to control features .

niallodowd, 2019-05-09:
Thanks for the response! However, I don't quite understand where to look
for all of the options for the camera object can you explain please? Also,
your example of camera.Gain=42 returns with an error
"LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line
16599)" Can you please elaborate on how I should set the parameters the
correct way? Also, camera.DecimationHorizontal=2 returns with the error
"object is not writable".

Thanks,
Niall

niallodowd, 2019-05-09:
I see now that I have to use camera.open() to change its parameters. Thanks!

niallodowd, 2019-05-09:
I'm having trouble extracting the actual value of parameters of the camera:

`cam.DecimationHorizontal` 

returns 

<pypylon.genicam.IInteger; proxy of <Swig Object of type 'GENAPI_NAMESPACE::IInteger *' at 0x000002534949D810> >

How should I extract the interger value of the decimation?

thiesmoeller, 2019-05-09:
The returned object is a genicam node.

The property

camera.DecimationHorizontal.Value

returns the current value

Other attributes are e.g. Min and Max

niallodowd, 2019-05-09:
Thank you! But I'm getting an error when I use that in my basic function:

`def getDeci():
    cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    cam.Open()
    decimation=cam.DecimationHorizontal.Value
    cam.close()
    return decimation`

There must be still something that I am missing

thiesmoeller, 2019-05-09:
... as we wanted to stay as close as possible to the C++ API we decided to
not follow pep8 for the pylon names.

It's can.Close not cam.close

thiesmoeller, 2019-05-09:
... and you should open the camera outer scope and just give the camera
object to your methods ...
Open/close every time is not efficient ..

Thies Möller <thies.moeller@googlemail.com> schrieb am Do., 9. Mai 2019,
21:05:

closed 2020-12-01 by thiesmoeller