basler/pypylon #110: Fail install in ubuntu16.04

pch9520, 2019-05-09:
ERROR: Could not find a version that satisfies the requirement pypylon==1.4.0 (from versions: none)
My python version is 3.5.2,
swig version is 4.0.0

thiesmoeller, 2019-05-09:
Ubuntu 16.04 on what  architecture ?
Error message from pip install ?
Swig is only required if you build from source ...

pch9520, 2019-05-10:
Ubuntu 16.04 on x86_64
Yes, Error message from pip install .

pch9520, 2019-05-10:
I have solved my problem, thank you again !

enesbedir1, 2019-05-24:
how did you solve this problem? can you write?

pch9520, 2019-06-04:
I install it by pypylon-1.4.0-cp35-cp35m-linux_x86_64.whl    
That's successful!

thiesmoeller, 2021-02-23:
Beginning with release 1.7.1 we have binary releases on all platforms, that you can install via pip.

update your pip version before by calling `python -m pip install --upgrade pip`

closed 2021-02-23 by thiesmoeller

szymonk211, 2021-06-30:
I had the same problem, for some reason ```python -m pip install pypylon``` is not working on Ubuntu 16.04.
Error message is:
```
ERROR: Could not find a version that satisfies the requirement pypylon (from versions: none)
ERROR: No matching distribution found for pypylon
```

```python -m pip install --upgrade pip``` didn't help.

I worked around the issue with command:
```pip install https://github.com/basler/pypylon/releases/download/1.5.4/pypylon-1.5.4-cp37-cp37m-linux_x86_64.whl```

Any newer version was not possible to install.

thiesmoeller, 2021-06-30:
See my comment above for solution.

szymonk211, 2021-06-30:
I used virtual env - interpreter was correct - upgrading pip didn't help.
Workaround is probably fine though - it's not a big problem for me.

thiesmoeller, 2021-07-02:
As this should work @szymonk211 , could you please call in your venv:

`python3 -m pip --version`

szymonk211, 2021-07-02:
pip 21.1.3

I use conda environment (it shouldn't matter). I don't have any problem installing pypylon on ubuntu 20 - only on ubuntu 16.

thiesmoeller, 2021-07-02:
Which conda version for us to reproduce?

szymonk211, 2021-07-03:
conda 4.8.3
The same problem is with pyenv 2.0.3

thiesmoeller, 2021-07-05:
checked: ubuntu 16 glibc is 2.23,
we now build pypylon version >= 1.7 with 2.24 glibc and are incompatible because of this.