basler/pypylon #111: What is the expected best latency?

ancabilloni, 2019-05-11:
I just tried this package especially with `opencv.py` script with a acA640-750uc. The best measure I could get for latency is at about 50ms. Is this possibly the best I could expect by using this package on Ubuntu (18.04 & 16.04)? I would like to find a way to shorten this latency even more if possible for a time sensitive application of measuring color spaces of image frame.

What else can I try to get better latency than 50ms? Really appreciate any input. Thanks!

P.S:
How I measure the latency is face the camera into a pre-recorded video on an ipad and compare the time difference between the ipad flash screen to my pc's image viewer with a timewatch. The scene was recorded in slow-mo to analyze.

thiesmoeller, 2019-05-11:
Hi @ancabilloni 
The measurement setup you are using is for glass2glass including also the rendering on your desktop and the possible multiple frame delay on your monitor.
From what you explain, it seemed to me that you want to do measurement 
on the video data. 

Additionally your current exposure time is part of the latency.

Highest performance is by using the OnImage callback.
grab strategies sample shows details ...

thiesmoeller, 2020-12-01:
No feedback by reporter,,,,

closed 2020-12-01 by thiesmoeller