basler/pypylon #112: Set ROI

argolomb, 2019-05-14:
When I try to set a ROI I get this error:

`camera.Width=2550`

genicam.AccessException: Node is not writable. : AccessException thrown in node 'Width' while calling 'Width.SetValue()' (file 'IntegerT.h', line 77)

but using pylon Viewer I can set ROI

There is any function to change the node to writable?

thiesmoeller, 2019-05-14:
Please call camera.Open() before first access to a feature. Examples are in
samples folder

closed 2020-12-01 by thiesmoeller

xiaoxstz, 2021-12-23:
I also came across a similar issue when I called `camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)`

> Could not apply configuration. Pylon::GenericException caught in OnOpened method msg=Node is not writable : AccessException thrown in node 'AcquisitionMode' while calling 'AcquisitionMode.FromString()' (file 'valuet.h', line 85) : RuntimeException thrown (file 'acquirecontinuousconfiguration.h', line 82)

But, this code is fine yesterday.

xiaoxstz, 2021-12-23:
It also occurs when I called `camera.Open()` after I added `camera.Open()`

The whole code  is shown below:
```python
from pypylon import pylon # pip install pypylon

class CamPylonWrapper:
    def __init__(self) -> None:
        # conecting to the first available camera
        self.camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
        self.camera.Open()

        # Grabing Continusely (video) with minimal delay
        self.camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

        self.__converter = pylon.ImageFormatConverter()

        # converting to opencv bgr format
        self.__converter.OutputPixelFormat = pylon.PixelType_BGR8packed
        self.__converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

        # get image size
        # self.width = self.camera.Width.GetValue()    # not right value
        # self.height = self.camera.Height.GetValue()
        ret,img = self.get_frame()
        self.width = img.shape[0]
        self.height = img.shape[1]

if __name__ == '__main__':
    camera = CamPylonWrapper()
```

xiaoxstz, 2021-12-23:
I solve it finally by the steps below:
* plug out the camera from PC or the power of camera
* wait for a while
* plug in again