basler/pypylon #114: Grabbing Multiple Images with Hardware Trigger

ruggiamp, 2019-05-20:
Hello,

I am trying to get some pictures with a acA2040-90umNIR using an external hardware trigger.
I manage to get one picture but when I have to take more than one I always get the timeout exception: 
'TimeoutException: Grab timed out. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1056)'
I always manage to get just one picture and no more. What am I doing wrong? I thought that after I call camera.StartGrabbing() the pictures are taken and sent to the buffer every time the camera receives a trigger (as long as the time distance between the triggers is long enough). Then camera.RetrieveResult() just retrieves them...
My code, basically is:

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.RegisterConfiguration(pylon.ConfigurationEventHandler(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete) 
camera.Open()
camera.AcquisitionMode.SetValue('SingleFrame')
camera.MaxNumBuffer = 15
camera.TriggerSelector.SetValue('FrameStart') 
camera.TriggerMode.SetValue('On')
camera.TriggerSource.SetValue('Line3') 
NumberOfPictures = 3
Images = []
camera.StartGrabbingMax(NumberOfPictures, pylon.GrabStrategy_OneByOne)

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(10000, pylon.TimeoutHandling_ThrowException)
    img = grabResult.Array
    Images.append(img) 

camera.Close()

thiesmoeller, 2019-05-27:
Hi,

If you use pylon viewer:
are you able to grab triggered images?

>

ruggiamp, 2019-05-27:
Hi,

Do you mean with hardware triggering?
I can acquire images using a software trigger pretty well. 
Is it posssible to hardware trigger with pylon viewer?

thiesmoeller, 2019-05-27:
;-) of course.
Just configure the same hardware and software configuration like
 in your final application.

And start image acquisition in pylon viewer. If your HW/SW setup works 
you should now see images arriving ..

ruggiamp, 2019-05-27:
OK,
yes, I managed. I had to press the 'continuous shot' button in pylon viewer toolbar to start the acquisition, but then it waits for hardware triggers if I set correctly the  'acquisition control'  parameters.
So, it is fine on Pylon Viewer I suppose...

thiesmoeller, 2019-05-27:
And you get an image with every HW trigger you apply ?

ruggiamp, 2019-05-28:
Yes. I also get one more when I start the triggering sequence and when I end it, but that I think it is the way I send signals with the trigger and I think I can solve it.
In any case, for any trigger signal I send I get an image.

ruggiamp, 2019-05-29:
Am I missing anything in my python code?

hahnec, 2019-10-23:
> I manage to get one picture but when I have to take more than one I always get the timeout exception: 'TimeoutException: Grab timed out. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1056)'

I suspect the issue that @ruggiamp faces lies in `camera.AcquisitionMode.SetValue('SingleFrame')` which is resolved when changed to `camera.AcquisitionMode.SetValue('Continuous')` as I then receive as many captures as defined in `NumberOfPictures`. Hope this helps.

ruggiamp, 2019-10-26:
Yep, I solved it doing that!
Thanks!

closed 2019-10-26 by ruggiamp