basler/pypylon #115: Set image width and height using python code

inislab, 2019-05-24:
Hi, I have a requirement to set image width and height to 800x600 . how can i achieve this using pypylon library. currently i am getting image dimension as 1280 x 1024 by default and i do not see any option to set custom image size

thiesmoeller, 2019-05-24:
All features that you can use in basler pylon viewer, or in the
documentation under docs.baslerweb.com are directly available in pypylon.

If the features name is Width.

You can access it via

camera.Width = 800

Please check that you called

camera.Open()

before

>

inislab, 2019-05-25:
Hi, as you said, i can able to change image size in pylon viewer but not able save my changes permanently. Values are going to default when i restart my system.

In docs.baslerweb.com, i do not see sample code for python to set image width/height

I am using acA1300-75gc camera

pault587, 2019-05-25:
After camera.Open()
camera.Width = 800
camera.Height = 600

thiesmoeller, 2020-12-01:
Look up Configuration set feature on docs.basler web.com

closed 2020-12-01 by thiesmoeller

sabrinaacardoso, 2022-04-27:
Hi, i have the same problem, and 

```camera.Width = 800
camera.Height = 600``` 

Doesn't work in my code.
Is there another way to change these values?

thiesmoeller, 2022-04-27:
Hi @sabrinaacardoso can you post a reduced code sample of what is not working ? And your camera model.

sabrinaacardoso, 2022-05-04:
In fact, I was able to use the

```
camera.Width
camera.Height
```

But only the upper left corner appears on the screen. 
To solve the problem I tried to use the function

```
camera.ImageCompressionMode( )
```

But I got an error.

Camera: a2A1920-51gcPRO

Code:
```
from pypylon import pylon
import cv2

## conecting to the first available camera
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

camera.ImageCompressionMode = "BaslerCompressionBeyond"
camera.ImageCompressionRateOption = "Lossless"
camera.BslImageCompressionRatio = 30.0

print("FPS:", camera.ResultingFrameRate.GetValue())

# Grabing Continusely (video) with minimal delay
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.imshow('title', img)

        if cv2.waitKey(1) & 0xFF == ord('q'):
            break
    grabResult.Release()
    
# Releasing the resource    
camera.StopGrabbing()

```

Error:
```
Traceback (most recent call last):
  File "/home/Camera_show.py", line 38, in <module>
    image = converter.Convert(grabResult)
  File "/home/.local/lib/python3.8/site-packages/pypylon/pylon.py", line 6895, in Convert
    return _pylon.ImageFormatConverter_Convert(self, *args)
_genicam.InvalidArgumentException: Cannot convert image. The passed source image is invalid. : InvalidArgumentException thrown (file 'ImageFormatConverter.cpp', line 77)
```

thiesmoeller, 2022-05-04:
Width And Height set the cropping of the video.
You have to also set OffsetX and OffsetY to move the cropped region around.

Just check  the documentation https://docs.baslerweb.com/image-roi.html#changing-position-and-size-of-an-image-roi

sabrinaacardoso, 2022-05-04:
I'll look at the documentation, thanks!