basler/pypylon #116: Segmentation fault when using two threads

sbraitbart, 2019-05-28:
Hello to all.
I am trying to grab from two cameras in free running mode using an acquisition thread per camera. After a few minutes of run time the script crashes with a Segmentation error. I tried a few grab strategies with the same result.
Am I doing something wrong?
I attach the code

Thanks
Shlomi
[BaslerAcuisition.txt](https://github.com/basler/pypylon/files/3226194/BaslerAcuisition.txt)

sbraitbart, 2019-05-28:
Forgot to mention:
I am using python 3.7 on Debian testing with pypylon 1.4 cp37 and pylon 5.2 
2x Basler USB3 acA1920-155um
Debian testing system 16GB kernel 4.19.0-5-amd64

thiesmoeller, 2019-05-31:
Hi Shlomi, 
while trying to reproduce this... can you post the backtrace while getting the segfault by running your program in gdb e.g.
`gdb -ex run --args <path_to_python3> BaslerAcquisition.txt`

On the first review, this looks ok, as you are just using python threading.

Regards,
Thies

sbraitbart, 2019-06-02:
Hi,
Thanks for the response. I run it using gdb as you instructed and the result below:

---------------------------------------------------------------------------------
Starting program: /usr/bin/python3 ./BaslerAcuisition.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1af0700 (LWP 7813)]
[New Thread 0x7ffff12ef700 (LWP 7814)]
[New Thread 0x7fffecaee700 (LWP 7815)]
[New Thread 0x7fffe7533700 (LWP 7816)]
[New Thread 0x7fffe6d32700 (LWP 7817)]
[New Thread 0x7fffe6531700 (LWP 7821)]
[New Thread 0x7fffe5d30700 (LWP 7822)]
[New Thread 0x7fffdfdcc700 (LWP 7823)]
[New Thread 0x7fffdf5cb700 (LWP 7824)]
[New Thread 0x7fffdedca700 (LWP 7825)]
[New Thread 0x7fffdcfcb700 (LWP 7826)]
[New Thread 0x7fffc3fff700 (LWP 7827)]
[New Thread 0x7fffc37fe700 (LWP 7828)]

Thread 13 "python3" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc3fff700 (LWP 7827)]
0x0000000000581c37 in PyUnicode_FromFormatV ()
---------------------------------------------------------------------------------------

sbraitbart, 2019-06-02:
I managed to improve the crash interval to crash after a longer period of time by using a single global Lock:
"
...
AcquisitionLock = threading.Lock()
...
              with AcquisitionLock:
                    Frame = AVAILABLE_CAMERAS[Cam].RetrieveResult(1000)
                    FID = Frame.GetID()
                    frm = Frame.GetArray()
                    Frame.Release()
...
"

sbraitbart, 2019-06-02:
a second run with gdb:

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1af0700 (LWP 9867)]
[New Thread 0x7ffff12ef700 (LWP 9868)]
[New Thread 0x7fffecaee700 (LWP 9869)]
[New Thread 0x7fffe7533700 (LWP 9870)]
[New Thread 0x7fffe6d32700 (LWP 9871)]
[New Thread 0x7fffe6531700 (LWP 9878)]
[New Thread 0x7fffe5d30700 (LWP 9879)]
[New Thread 0x7fffdfdcc700 (LWP 9880)]
[New Thread 0x7fffdf5cb700 (LWP 9881)]
[New Thread 0x7fffdedca700 (LWP 9882)]
[New Thread 0x7fffdcfcb700 (LWP 9883)]
[New Thread 0x7fffc3fff700 (LWP 9884)]
[New Thread 0x7fffc37fe700 (LWP 9885)]

Thread 13 "python3" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc3fff700 (LWP 9884)]
0x0000000000598587 in PyType_GenericAlloc ()

sbraitbart, 2019-06-02:
and a third run: (please let me know if you need me to perform other tests)

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff1af0700 (LWP 10180)]
[New Thread 0x7fffef2ef700 (LWP 10181)]
[New Thread 0x7fffecaee700 (LWP 10182)]
[New Thread 0x7fffe7533700 (LWP 10183)]
[New Thread 0x7fffe6d32700 (LWP 10184)]
[New Thread 0x7fffe6531700 (LWP 10189)]
[New Thread 0x7fffe5d30700 (LWP 10190)]
[New Thread 0x7fffdfdcc700 (LWP 10191)]
[New Thread 0x7fffdf5cb700 (LWP 10192)]
[New Thread 0x7fffdedca700 (LWP 10193)]
[New Thread 0x7fffdcfcb700 (LWP 10194)]
[New Thread 0x7fffc3fff700 (LWP 10195)]
[New Thread 0x7fffc37fe700 (LWP 10196)]

Thread 14 "python3" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc37fe700 (LWP 10196)]
0x00000000005a37c3 in PyObject_Malloc ()

sbraitbart, 2019-06-02:
Hi
I added a small sleep period at the end of the execution loop of the thread. It seemed to improve stability even more.

sbraitbart, 2019-06-03:
I have a work around which seems to be stable (more than 2 hours with no crash).
It is to add a small sleep at the end of the execution loop and a global thread Lock for addressing the camera.
the attached code works. 
[BaslerAcuisition_workaround.txt](https://github.com/basler/pypylon/files/3246679/BaslerAcuisition_workaround.txt)

thiesmoeller, 2019-06-05:
@sbraitbart : we identified an incorrect handling of the GIL as the root cause to your problem in GetBuffer SWIG implementation... already testing the fix

driesFM, 2019-06-07:
I am experiencing a similar issue:

- I have the latest pylon Suite installed, with pypylon 1.4.0 (installed using pip).
- I have two Python threads running
- I have two Basler GigE cameras connected to the PC
- Each thread is grabbing images and processing them

After some time (10min - 2h), I am getting a Python crash. The crash dump shows an _access_violation_. I still need to trace it down exactly where it is happening, but it seems to related to the image grabbing.

I will have a look into the workaround.

thiesmoeller, 2019-06-07:
... as releasing a full update with the fix might take some time:

here is the patch:
[fix_pypylon_memory_corruption.txt](https://github.com/basler/pypylon/files/3267213/fix_pypylon_memory_corruption.txt)

driesFM, 2019-06-10:
Thank you for sharing the fix. I will still need to try it.
As an additional test, I was running my code with only one camera. But I am also getting an access violation exception.

In my application, I have a dedicated thread for the pylon interface (see attached code). Unfortunately, this thread throwed again an access violation, even when I was using only a single camera.
I have attached my Python code, including a screenshot of the exception that was captured (from a crash dump).

Will your patch also fix this exception, or is this something else?

[basler_cameraReader.txt](https://github.com/basler/pypylon/files/3272133/basler_cameraReader.txt)
[image]

thiesmoeller, 2019-06-10:
Hi @driesFM ,

pretty sure, that it will fix your case too. The memory allocation, that happens in PyByte_from_StringAndSize is exactly the point, where we didn't handle the GIL correctly, and this will/could lead to a segfault, if you are running a multithreaded python application.

Most applications are using the background threads of the pylon grab-engine to do the frame processing on multiple cameras, and just the main thread for GUI etc.
This is why this issue was able to survive in the code for such a long time ....

Starting with python 3.6 it is now possible to find the issue [ you can do this on your test program too ] with pylon debug support, that is included into every interpreter release:

the environment variable `PYTHONMALLOC` controls this debug mechanisms.
So under Linux you can just write

`$ PYTHONMALLOC=malloc_debug   python3 sample_script.py`

which will immediately show the issue:
```

Fatal Python error: Python memory allocator called without holding the GIL

Thread 0x00007f661b7fe700 (most recent call first):
  File "sample_script.py", line 44 in run
  File "/home/tmoeller/.pyenv/versions/3.6.5/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/home/tmoeller/.pyenv/versions/3.6.5/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f661bfff700 (most recent call first):
  File "/home/tmoeller/.pyenv/versions/3.6.5/lib/python3.6/site-packages/pypylon-1.4.0+pylon5.0.12.dbg-py3.6-linux-x86_64.egg/pypylon/pylon.py", line 2020 in GetBuffer
  File "/home/tmoeller/.pyenv/versions/3.6.5/lib/python3.6/site-packages/pypylon-1.4.0+pylon5.0.12.dbg-py3.6-linux-x86_64.egg/pypylon/pylon.py", line 1600 in GetArray
  File "/home/tmoeller/.pyenv/versions/3.6.5/lib/python3.6/site-packages/pypylon-1.4.0+pylon5.0.12.dbg-py3.6-linux-x86_64.egg/pypylon/pylon.py", line 150 in func_wrapper
  File "sample_script.py", line 59 in run
  File "/home/tmoeller/.pyenv/versions/3.6.5/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/home/tmoeller/.pyenv/versions/3.6.5/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f6649607700 (most recent call first):
  File "/home/tmoeller/.pyenv/versions/3.6.5/lib/python3.6/site-packages/pypylon-1.4.0+pylon5.0.12.dbg-py3.6-linux-x86_64.egg/pypylon/pylon.py", line 3007 in IsGrabbing
  File "sample_script.py", line 107 in start_acuisition
  File "sample_script.py", line 111 in <module>
Aborted (core dumped)

```

sbraitbart, 2019-06-11:
Thank you very much for addressing this issue so fast.
Can you please elaborate on how to apply the patch, since I haven't done this before. (linux system)
Thanks 
Shlomi

thiesmoeller, 2019-06-11:
enter project directory
`cd pypylon`

apply patch
`patch -p1 < fix_pypylon_memory_corruption.txt`

and build according to [README](https://github.com/basler/pypylon/blob/master/README.md)

sbraitbart, 2019-06-12:
Thanks
It is stable now.

closed 2019-06-13 by sbraitbart

driesFM, 2019-06-14:
I can also confirm that the fix is working. My program is already running stable for 3 days, while before I couldn't reach 3 hours. Thank you for providing the fix so quickly!

ridvanozdemir, 2023-08-11:
how can I apply this patch to windows o.s.?

I also have similiar problem:

Windows fatal exception: access violation

Thread 0x00000d34 (most recent call first):
  File "C:\Users\Kadir\AppData\Roaming\Python\Python310\site-packages\pypylon\pylon.py", line 3495 in RetrieveResult
  File "c:\ecg\ecg01_3d_2d_v28.py", line 1277 in run

Thread 0x00002264 (most recent call first):
  File "C:\Users\Kadir\AppData\Roaming\Python\Python310\site-packages\pypylon\pylon.py", line 3495 in RetrieveResult
  File "c:\ecg\ecg01_3d_2d_v28.py", line 1277 in run

Main thread:
Current thread 0x000020f8 (most recent call first):
  File "c:\ecg\ecg01_3d_2d_v28.py", line 959 in ImageUpdateSlot
  File "c:\ecg\ecg01_3d_2d_v28.py", line 1412 in <module>
  File "C:\ProgramData\anaconda3\lib\site-packages\spyder_kernels\py3compat.py", line 356 in compat_exec
  File "C:\ProgramData\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 469 in exec_code
  File "C:\ProgramData\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 611 in _exec_file
  File "C:\ProgramData\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 524 in runfile
  File "C:\Users\Kadir\AppData\Local\Temp\ipykernel_10752\365309949.py", line 1 in <module>

Restarting kernel...