basler/pypylon #117: Is it possible to parameterize parameters, save them in a file and load them each time the camera is accessed?

zamarjavier, 2019-05-29:
Hello,

I want to fix some values on my Basler Dart camera. The idea is to save those parameters in a file and each time the camera is initialized, load those parameters by defoult.

Thank you so much.

thiesmoeller, 2019-06-01:
If you want to do this without any programming.
Just open your device with pylon viewer. Change values as required and follow the documentation on Usersets.
They are stored in camera flash memory,
[user sets](https://docs.baslerweb.com/#t=en%2Fuser_sets.htm)

thiesmoeller, 2019-06-01:
Programmatic way ... See this example

[Load save test](https://github.com/basler/pypylon/blob/master/tests/pylon_tests/emulated/loadandsavetest.py)

closed 2020-12-01 by thiesmoeller