basler/pypylon #119: How to use pypylon API to program for "grabbing an image with hardware trigger"

cute-tiger, 2019-06-04:
After extensive searching and reading related Issues and examples, I am still not able to figure out the specific commands and sequence to program using Pypylon API to perform task: 
1. set up the camera to be hardware trigger mode
2. once there is a signal from sensor, grab an image from camera and save it to a file
3. repeat step 2 for ever

Anyone can share detailed scripts using pypylon API?

MaximilianHoffmann, 2019-06-04:
Something like this?
```
cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    cam.Open()
    cam.MaxNumBuffer = 200
    cam.ExposureAuto.SetValue('Off')
    #
    cam.ExposureTime.SetValue(exposure_time_to_set)
    # Set Gain to zero and turn of exposure auto
    cam.GainAuto.SetValue('Off')
    cam.Gain.SetValue(0)
    # Set Trigger Mode On
      cam.TriggerMode.SetValue('On')
        #cam.AcquisitionFrameRateEnable.SetValue('Off')
    cam.TriggerDelay.SetValue(0)
    cam.TriggerSelector.SetValue('FrameStart')
    cam.TriggerSource.SetValue('Line3')
    cam.TriggerActivation.SetValue('RisingEdge')
    # Trigger On Line 3 FrameStart Rising Edge, Exposure Out Line 2.
    cam.LineSelector.SetValue('Line4')
    cam.LineInverter.SetValue('On')
    cam.LineMode.SetValue('Output')
    cam.LineSource.SetValue('ExposureActive')
    cam.PixelFormat.SetValue('Mono8')

```

And the grab in a while loop with a good timout exception handling?

pault587, 2019-06-04:
Once the camera is sending images after triggering it isn’t any different than if it was streaming. Other than you may timeout if you haven’t triggered the camera in a while, be sure to set that parameter appropriately. Please refer to the sample code for grabbing images (streaming or triggered the code doesn’t care). In the samples it even includes doing this in a loop. 

If you don’t know how to set the cameras trigger mode that is a quick search in the manual, I did that for you here…

https://docs.baslerweb.com/#t=en%2Ftrigger_selector.htm&rhsearch=trigger&rhhlterm=trigger

“Hardware Triggering

If your camera supports hardware triggering, an electrical signal applied to one of the camera's input lines can act as a trigger signal.

To configure the camera for hardware triggering:

Select the trigger type that you want to configure, e.g., FrameStart.
Set the trigger mode of the selected trigger to On.
Set the trigger source to one of the available hardware trigger sources.
If desired, use the Trigger Activation and the Trigger Delay features to further configure hardware triggering.
Repeat the steps above for all trigger types that you want to control with an electrical signal.
Now, you can trigger the camera by applying electrical signals to the selected input line.

“

You can even set all the parameters in the Viewer and the camera will remember the settings until powered off.  Later you load these from a saved file or do it programmatically as stated in other posts.

Are you having a specific issue with a feature? What are you asking for seems to be well documented. Which part is not working for you?

cute-tiger, 2019-06-05:
## Here is my code:
## Tested on Basler acA640-120gm (linux 64bit , python 3.5)

from pypylon import pylon
import cv2
import numpy

# connecting to the first available camera
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

camera.Open()

# To see which device info is available 
print("DeviceClass: ", camera.GetDeviceInfo().GetDeviceClass())
print("DeviceFactory: ", camera.GetDeviceInfo().GetDeviceFactory())
print("ModelName: ", camera.GetDeviceInfo().GetModelName())

############################################################
Hardware_Trigger = False 

if Hardware_Trigger:
    # reset registration 
    camera.RegisterConfiguration(pylon.ConfigurationEventHandler(), pylon.RegistrationMode_ReplaceAll,     pylon.Cleanup_Delete)

    # The parameter MaxNumBuffer can be used to control the count of buffers
    # allocated for grabbing. The default value of this parameter is 10.
    camera.MaxNumBuffer = 5

    # set exposure time
    camera.ExposureTimeRaw.SetValue(100)

    # Select the Frame Start trigger 
    camera.TriggerSelector.SetValue('FrameStart')
    # Acquisition mode
    camera.AcquisitionMode.SetValue('SingleFrame')
    # Enable triggered image acquisition for the Frame Start trigger 
    camera.TriggerMode.SetValue('On') 
    # Set the trigger source to Line 1 
    camera.TriggerSource.SetValue('Line1')
    # Set the trigger activation mode to rising edge 
    camera.TriggerActivation.SetValue('RisingEdge')
    # Set the delay for the frame start trigger to 300 µs 
    camera.TriggerDelayAbs.SetValue(300.0)
    # Pixel format
    camera.PixelFormat.SetValue('Mono8')

##############################################################

camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

count = 1
while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        print("yes, an image is grabbed successfully")

        cv2.imwrite('save_images/%06d.png'%count, img)
        print("image/%06d.png saved"%count)
        count += 1
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()
    
# Releasing the resource    
camera.StopGrabbing()

# This is end of code
##########################################################

If set Hardware_Trigger = False , the result is
.....
yes, an image is grabbed successfully
image/000242.png saved
yes, an image is grabbed successfully
image/000243.png saved
.....

However, if set Hardware_Trigger = True , there is no response to the photoeye sensor trigger, and after 5 seconds, it gives error:
return _pylon.InstantCamera_RetrieveResult(self, *args)
_genicam.TimeoutException: Grab timed out. The acquisition is not started. In 'SingleFrame' acquisition mode the acquisition has to be started for every frame, see also GrabOne(). : TimeoutException thrown (file 'InstantCameraImpl.h', line 1026)

Can anyone help?  I don't understand why the sensor signal does not make grabbing successful.

MaximilianHoffmann, 2019-06-05:
What is the HIGH voltage of your trigger sensor and did you measure it's presence with an oscilloscope or similar?

cute-tiger, 2019-06-05:
It should be about 15 watt. I tested in Pylon Viewer using with trigger sensor, it worked well.

thiesmoeller, 2019-06-05:
One answer is in the error message.
SingleFrame is only if you want to automatically stop the acquisition after
one frame ....
So you want Continuous acquisition mode ...

This is what you get in pylon viewer too...

Ist your full triggered setup working in pylon viewer ?

cute-tiger, 2019-06-05:
Hi thiesmoeller, I commented out the line "camera.AcquisitionMode.SetValue('SingleFrame')", and modified the last part of code as below:
=====================================================
count_trigger = 1
count_image =1

time_0 = time.time()

while camera.IsGrabbing():
    print("count_trigger: ", count_trigger)
    count_trigger += 1
    
    grabResult = camera.RetrieveResult(10000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        print("grabResult is succeeded!")
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.imwrite('save_images/%06d.png'%count_image, img)
        print("%06d.png saved"%count_image)
        count_image += 1

    grabResult.Release()
    
# Releasing the resource    
camera.StopGrabbing()
==========================================================
Then the results are like below.  
DeviceClass:  BaslerGigE
DeviceFactory:  GigE/BaslerGigE 5.2.0.13457
ModelName:  acA640-120gm
count_trigger:  1
count_trigger:  2
count_trigger:  3
count_trigger:  4
count_trigger:  5
grabResult is succeeded!
000001.png saved
count_trigger:  6
grabResult is succeeded!
000002.png saved
count_trigger:  7
count_trigger:  8
count_trigger:  9
count_trigger:  10
count_trigger:  11
count_trigger:  12
count_trigger:  13
count_trigger:  14
count_trigger:  15
count_trigger:  16
count_trigger:  17
count_trigger:  18
count_trigger:  19
grabResult is succeeded!
000003.png saved
count_trigger:  20
....

Every time when the sensor is triggered, it outputs "count_trigger: *", but after I triggered the sensor 19 times, there are only 3 images saved.  How to ensure the image is saved for each trigger? Is this due to buffer setting or something else?

cute-tiger, 2019-06-05:
When I was debugging the root cause, I found when use Pylon Viewer continuous shot  to check the camera, if not using hardware trigger mode, the camera works fine. But when I set the hardware trigger mode "On", in the message window in Pylon Viewer, it gives me this error message: 

Type	Time	Source	Message
Error	2019-06-05 15:18:44.175	Basler acA1920-40gm (23014018)	Image acquisition on "Basler acA1920-40gm (23014018)" failed! Error: 0xe1000014 "The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size."

Do anyone know what is root cause to this error and how to fix it?

thiesmoeller, 2019-06-05:
This was exactly my first idea after your last mail ;-)

You are probably overtriggering the device.
Do you have an oscilloscopes to check your trigger input ?
Is the frequency below the current max fps of your device ?

Check the feature
https://docs.baslerweb.com/#t=en%2Fresulting_frame_rate.htm feature

cute-tiger, 2019-06-05:
Hi thiesmoeller, the frequency of trigger is much lower than the max fps of the camera. 
is there any other setting contradictory to hardware trigger that leads to this issue?

cute-tiger, 2019-06-05:
The issue has been solved by changing "packet size" to 1500 from 600 in Pylon Viewer.

cute-tiger, 2019-06-05:
I am closing this Issue now. Thanks to everyone who helped.

closed 2019-06-05 by cute-tiger

thiesmoeller, 2019-06-05:
Last note:
Use the bandwidth optimizer available in pylon viewer to get optimal
settings in free running mode before going to triggered mode

RCC-GIT, 2020-05-19:
Hello, I am trying to get an image every hardware rising edge trigger . Tiggering can be evry  200ms to every hour .  I am using this code but always get a time out Error from camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) . If fast enough I can trigger an image . Is there a wait to loop and trigger only on harware trigger ? 
AcquisitionMode.SetValue('SingleFrame') or AcquisitionMode.SetValue('Continuous') does not change anything
Trigger tested on Pylonwiever and works OK 
Here under is the code
Thank you for you help .
`
# mise en route de la camera et configuration 
# =====================================================
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
print("DeviceClass: ", camera.GetDeviceInfo().GetDeviceClass()) 
print("DeviceFactory: ", camera.GetDeviceInfo().GetDeviceFactory()) 
print("ModelName: ", camera.GetDeviceInfo().GetModelName())

# The parameter MaxNumBuffer can be used to control the count of buffers
# allocated for grabbing. The default value of this parameter is 10.
camera.MaxNumBuffer = 5

# set exposure time
#camera.ExposureTimeRaw.SetValue(100)

# Select the Frame Start trigger 
camera.TriggerSelector.SetValue('FrameStart')
# Acquisition mode
camera.AcquisitionMode.SetValue('SingleFrame')
# Enable triggered image acquisition for the Frame Start trigger 
camera.TriggerMode.SetValue('On') 
# Set the trigger source to Line 1 
camera.TriggerSource.SetValue('Line1')
# Set the trigger activation mode to rising edge 
camera.TriggerActivation.SetValue('RisingEdge')
# Set the delay for the frame start trigger to 300 µs 
camera.TriggerDelayAbs.SetValue(300.0)
# Pixel format
camera.PixelFormat.SetValue('Mono8')

# demarrage acquisition
# =====================================================
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
converter = pylon.ImageFormatConverter()

# convertion openCV BGR format 
converter.OutputPixelFormat = pylon.PixelType_BGR8packed 
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

count = 1 
while camera.IsGrabbing(): 
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
  #  camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 

    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        print("yes, an image is grabbed successfully")

        cv2.imwrite('save_images/%06d.png'%count, img)
        print("image/%06d.png saved"%count)
        count += 1
        k = cv2.waitKey(1)
        if k == 27:
            break
    
    grabResult.Release()

camera.StopGrabbing()
camera.Close()
`

RCC-GIT, 2020-05-19:
Again , possible to have a test as follows in a loop 

If camera is triggering:
   grab ....

shenjun1994, 2020-06-09:
I am using the camera in the hardware trigger mode, could you show me your complete code about it, thanks!