basler/pypylon #120: Ways to maximize frame rate

keven425, 2019-06-12:
This is a question, not an issue.

We currently grab using `pylon.GrabStrategy_OneByOne`, and software trigger. We have to convert the images to numpy array for processing. We currently use `converter.Convert(result).GetArray()` to get `PixelType_BGR8packed` and `OutputBitAlignment_MsbAligned`.

Our camera (`Basler acA2440-75uc`) can theretically reach 75 fps. We can reach only 40 fps with above set up. If we remove converter step, we can reach 70 fps. We will be happy with 70 fps.

How could we achieve that, by still converting images to numpy array? We are OK with async.

thiesmoeller, 2019-06-13:
If you don't need the image information of frame N before triggering frame N+1,
You should trigger between retrieveResult and the conversion step.
This will pipeline the two operations.
While the conversion runs, the acquisition,readout and transmission runs on the camera

keven425, 2019-06-13:
Got it. Is this a good pattern for pipelining the operations? https://github.com/basler/pypylon/blob/master/samples/grabusinggrabloopthread.py

Or is there a more efficient way to do so?

thiesmoeller, 2019-06-13:
This is the most efficient way.
It uses the pylon internal grab thread.

closed 2019-06-20 by keven425