basler/pypylon #122: no module named

kangjieHong, 2019-06-13:


thiesmoeller, 2019-06-13:
base error is:
`ImportError: libpython3.6m.so.1.0: cannot open shared object file: No such file or directory`

are you running in some not-activated venv?
can try to test, that it is an environment error, by explicitly setting LD_LIBRARY_PATH to `libpython3.6m.so.1.0` directory

ggrrin, 2019-08-06:
Hi, is there a solutions? I have the same error for python 3.7. I ve installed the package successuflly as follows:
`pip install pypylon-1.4.0-cp37-cp37m-linux_x86_64.whl`

But got similar error:
`Python 3.7.3 (default, Mar 27 2019, 22:11:17) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pypylon import pylon
Traceback (most recent call last):
  File "/home/nuc/miniconda3/lib/python3.7/site-packages/pypylon/pylon.py", line 42, in swig_import_helper
    return importlib.import_module(mname)
  File "/home/nuc/miniconda3/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: libpython3.7m.so.1.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/nuc/miniconda3/lib/python3.7/site-packages/pypylon/pylon.py", line 45, in <module>
    _pylon = swig_import_helper()
  File "/home/nuc/miniconda3/lib/python3.7/site-packages/pypylon/pylon.py", line 44, in swig_import_helper
    return importlib.import_module('_pylon')
  File "/home/nuc/miniconda3/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_pylon'
`
I am using miniconda python 3.7.3 with default environment and the missing file is located in 
`/home/nuc/miniconda3/lib`. Can I somehow workaround the issue?

listing the directory:
`(base) nuc@nuc:~/miniconda3/lib$ ls -l | grep libpython3.7m.so.1.0
lrwxrwxrwx  1 nuc nuc       20 kvě 20 14:50 libpython3.7m.so -> libpython3.7m.so.1.0
lrwxrwxrwx  1 nuc nuc       20 kvě 20 14:50 libpython3.7m.so.1 -> libpython3.7m.so.1.0
-rwxrwxr-x  1 nuc nuc  3797488 kvě 20 14:50 libpython3.7m.so.1.0
`

edwardmbsmith, 2019-09-05:
Are python development tools installed, e.g. `sudo yum install python36-devel`?

thiesmoeller, 2019-09-05:
the way we link doesn't work in anaconda environment. We'll look into changing this
either use LD_LIBRARY_PATH:

LD_LIBRARY_PATH=$HOME/miniconda3/lib python ....

or just recompile the project in your environment. ( with pylon5 and swig in your path according to readme )

```
(anaconda3-5.2.0) :~$ pip install git+https://github.com/basler/pypylon.git
Collecting git+https://github.com/basler/pypylon.git
  Cloning https://github.com/basler/pypylon.git to /tmp/pip-req-build-y_owalh7
  Running command git clone -q https://github.com/basler/pypylon.git /tmp/pip-req-build-y_owalh7
Building wheels for collected packages: pypylon
  Building wheel for pypylon (setup.py) ... done
  Created wheel for pypylon: filename=pypylon-1.4.0+pylon5.0.12-cp36-cp36m-linux_x86_64.whl size=3693775 sha256=7b3e822f8bd306a78d81d411e4da49108620b89de42fc6e1cae8b1e404abac38
  Stored in directory: /tmp/pip-ephem-wheel-cache-hwlbu8sp/wheels/6a/ea/fa/8a8ba2fb1654e90c0f5a1c7c218ffb0a6e68eb9a44c01df6ba
Successfully built pypylon
Installing collected packages: pypylon
Successfully installed pypylon-1.4.0+pylon5.0.12
(anaconda3-5.2.0) :~$ python
Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
[GCC 7.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pypylon.pylon as py
>>> py.__version__
'1.4.0+pylon5.0.12'

```

lgribeiro, 2020-01-09:
I have ubuntu 16.04  and envirioment  with python3.6 
I got the same error but I fixed it :)
I run:
sudo add-apt-repository ppa:deadsnakes/ppa 
sudo apt install python3.6-dev

AStransients, 2020-03-25:
I have the same problem. "No module named 'pypylon'"
Running WIN10 64bit and python 3.6 with anaconda.

Anybody has an approved fix?