basler/pypylon #123: cant access basler camera from docker container

kumarkumaresh, 2019-06-17:
i have created c++ application and i am calling pylon apis to connect and control basler camera(acA 1920-25gc).
I have created a deb package for linux and it accesses camera successfully when i run it locally as an application, but when i containerize the same exe. it runs as a container but not able to access the camera.

matt4lieb, 2019-06-17:
You have to mount your usb devices into your docker.

kumarkumaresh, 2019-06-18:
Thanks for reply but we have connected Basler camera through RJ45 not with usb .

matt4lieb, 2019-06-18:
Shouldn't be a issue. You can use the tag --device and pass-through the device to the container.

kumarkumaresh, 2019-06-18:
while starting the container from docker image i have already tried the following command but it didn't work.
docker run -it --device=/dev:/dev  -p 5555:5555 <imagename>

 also played around with /dev/snd etc.
but i am not sure where my device is .
/dev has following files and folder. how do i know where my device is mounted.

autofs         cpu_dma_latency  hidraw1    loop-control        null   pts     sdb1      tty    tty16  tty24  tty32  tty40  tty49  tty57  tty8     usb   vcsa1        vhost-net
block          cuse             hidraw2    mapper              port   random  sg0       tty0   tty17  tty25  tty33  tty41  tty5   tty58  tty9     vcs   vcsa2        watchdog
bsg            disk             hpet       mcelog              ppp    rtc     sg1       tty1   tty18  tty26  tty34  tty42  tty50  tty59  ttyS0    vcs1  vcsa3        watchdog0
btrfs-control  dri              hugepages  mem                 pps0   rtc0    shm       tty10  tty19  tty27  tty35  tty43  tty51  tty6   ttyS1    vcs2  vcsa4        zero
bus            fb0              initctl    memory_bandwidth    pps1   sda     snapshot  tty11  tty2   tty28  tty36  tty44  tty52  tty60  ttyS2    vcs3  vcsa5
char           fd               input      mqueue              psaux  sda1    snd       tty12  tty20  tty29  tty37  tty45  tty53  tty61  ttyS3    vcs4  vcsa6
console        full             kmsg       net                 ptmx   sda2    stderr    tty13  tty21  tty3   tty38  tty46  tty54  tty62  uhid     vcs5  vfio
core           fuse             kvm        network_latency     ptp0   sda3    stdin     tty14  tty22  tty30  tty39  tty47  tty55  tty63  uinput   vcs6  vga_arbiter
cpu            hidraw0          log        network_throughput  ptp1   sdb     stdout    tty15  tty23  tty31  tty4   tty48  tty56  tty7   urandom  vcsa  vhci

thiesmoeller, 2019-06-18:
[docker networking setup](https://docs.docker.com/config/containers/container-networking/)

matt4lieb, 2019-06-25:
@kumarkumaresh , did you found a solution?

kumarkumaresh, 2019-06-25:
Yes, I changed network of my container to host. Then the container was able
to access the camera.

gustavoponce7, 2019-06-28:
Hi, I am facing the same issue when I try to see the camera from my docker container, but in my case I am using the **USB port**, any suggestion?

Configuration:
- MacBook Pro (15-inch, 2018)
- Camera serial number: 22945247
- Camera model: daa2500-14um

**Docker-compose file**
version: "3"

services:
    Cameraserver-test:
        privileged: true
        build:
            context: .
            dockerfile: Dockerfile
        container_name: cameraserver-test
        volumes:
            - .:/cameraserver
        ports:
            - "7105:7105"
        command:
            - /bin/bash
            - -c
            - |
                udevadm control --reload-rules
                python3.6 -V
                pip3 -V
                python3.6 /cameraserver/server.py
        devices:
            - "/dev/:/dev/"

**ls /dev**
afsc_type5                  ptypc                       ptyt2                       ptyw8                       ttyqe                       ttytf
auditpipe                   ptypd                       ptyt3                       ptyw9                       ttyqf                       ttyu0
auditsessions               ptype                       ptyt4                       ptywa                       ttyr0                       ttyu1
autofs                      ptypf                       ptyt5                       ptywb                       ttyr1                       ttyu2
autofs_control              ptyq0                       ptyt6                       ptywc                       ttyr2                       ttyu3
autofs_homedirmounter       ptyq1                       ptyt7                       ptywd                       ttyr3                       ttyu4
autofs_notrigger            ptyq2                       ptyt8                       ptywe                       ttyr4                       ttyu5
autofs_nowait               ptyq3                       ptyt9                       ptywf                       ttyr5                       ttyu6
bpf0                        ptyq4                       ptyta                       random                      ttyr6                       ttyu7
bpf1                        ptyq5                       ptytb                       rdisk0                      ttyr7                       ttyu8
bpf2                        ptyq6                       ptytc                       rdisk0s1                    ttyr8                       ttyu9
bpf3                        ptyq7                       ptytd                       rdisk0s2                    ttyr9                       ttyua
bpf4                        ptyq8                       ptyte                       rdisk1                      ttyra                       ttyub
console                     ptyq9                       ptytf                       rdisk1s1                    ttyrb                       ttyuc
cu.Bluetooth-Incoming-Port  ptyqa                       ptyu0                       rdisk1s2                    ttyrc                       ttyud
disk0                       ptyqb                       ptyu1                       rdisk1s3                    ttyrd                       ttyue
disk0s1                     ptyqc                       ptyu2                       rdisk1s4                    ttyre                       ttyuf
disk0s2                     ptyqd                       ptyu3                       sdt                         ttyrf                       ttyv0
disk1                       ptyqe                       ptyu4                       stderr                      ttys0                       ttyv1
disk1s1                     ptyqf                       ptyu5                       stdin                       ttys000                     ttyv2
disk1s2                     ptyr0                       ptyu6                       stdout                      ttys001                     ttyv3
disk1s3                     ptyr1                       ptyu7                       systrace                    ttys002                     ttyv4
disk1s4                     ptyr2                       ptyu8                       tty                         ttys003                     ttyv5
dtrace                      ptyr3                       ptyu9                       tty.Bluetooth-Incoming-Port ttys004                     ttyv6
dtracehelper                ptyr4                       ptyua                       ttyp0                       ttys1                       ttyv7
fbt                         ptyr5                       ptyub                       ttyp1                       ttys2                       ttyv8
fd                          ptyr6                       ptyuc                       ttyp2                       ttys3                       ttyv9
fsevents                    ptyr7                       ptyud                       ttyp3                       ttys4                       ttyva
io8log                      ptyr8                       ptyue                       ttyp4                       ttys5                       ttyvb
io8logmt                    ptyr9                       ptyuf                       ttyp5                       ttys6                       ttyvc
io8logtemp                  ptyra                       ptyv0                       ttyp6                       ttys7                       ttyvd
[... cut ...]
ptyp7                       ptysd                       ptyw3                       ttyq9                       ttyta                       urandom
ptyp8                       ptyse                       ptyw4                       ttyqa                       ttytb                       xcpm
ptyp9                       ptysf                       ptyw5                       ttyqb                       ttytc                       zero
ptypa                       ptyt0                       ptyw6                       ttyqc                       ttytd
ptypb                       ptyt1

kumarkumaresh, 2019-06-29:
First try running docker container from image in command line and include
-- network host, this should work.
Then you can include this network in your compose file and run.

gustavoponce7, 2019-07-01:
I have included the "--network host" as argument but still not able to see the camera from the docker container.

I just realized that "pylon IP Configurator" is not displaying the camera when is connected, so I was wondering if this can be related to the camera visibility in the docker container?

[image]

kumarkumaresh, 2019-07-02:
Yes it should be visible in the pylon ip configurator first,in your case
first set some static ip to the device or make the auto ip set and
configure the camera first. Click some pics from the pylon viewer app first.

gustavoponce7, 2019-07-02:
@kumarkumaresh  thanks for the information but based on @thiesmoeller [feedback](https://github.com/basler/pypylon/issues/133#issuecomment-507755361) my camera(daa2500-14um) uses a host/device architecture and doesn't need a network layer to communicate.

So I would like to know if there is another needed configuration in order to see the camera in my docker container.

I have tried with **--network host** and **--device=/dev:/dev** but the docker is not able to access the camera

Any suggestions would be greatly appreciated

nicolaifsf, 2019-07-16:
Hi @gustavoponce7, not sure what OS you're running Docker on, but I've been looking into a similar issue I was having on my mac and found https://github.com/docker/for-mac/issues/2716 which seems to indicate that `--network host` doesn't work for Mac without some work around.

May be irrelevant to the issue you are having, but figured I'd mention it in case it does be of some help.

Cheers!

gustavoponce7, 2019-08-15:
At the end we decided to migrate from MAC to Linux(Ubuntu 18.04 LTS) and now we are able to see the camera from the docker container using the next command

**docker run --privileged <CONTAINER-NAME>**

Here a possible solution for MAC (I didn't test it)
https://milad.ai/docker/2018/05/06/access-usb-devices-in-container-in-mac.html

closed 2020-12-01 by thiesmoeller

IgnacioEscorz, 2022-11-17:
> Yes, I changed network of my container to host. Then the container was able to access the camera.
> […](#)

Hello @kumarkumaresh 
Im facing the same problem. How wxactly did you do that? I tried port mapping 3956:3956 alone, with upd and tcp. Network default existing, and network_mode: "host" and it didnt worked. Im using docker compose up on a yml 3.8 version. Thank you

Stefanoo9600, 2022-12-22:
Hi @IgnacioEscorz ,
have you solved the problem? I have a similar problem. 
Thank you in advance

IgnacioEscorz, 2022-12-23:
> Hi @IgnacioEscorz , have you solved the problem? I have a similar problem. Thank you in advance

Hi! Yes, try with the following.
You just need to fill the information for Destination Port and IP address and make sure you have those declared in the docker files.
Let me know if that worked.

from pypylon import pylon
import cv2

ip_address = 'XXX.XXX.XXX.XXX'
destination_ip_address = 'XXX.XXX.XXX.XXX'
factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = ptl.CreateDeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = factory.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)

camera.Open()
camera.StreamGrabber.DestinationPort = XXXXX

# conecting to the first available camera
# Grabing Continusely (video) with minimal delay
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.namedWindow('title', cv2.WINDOW_NORMAL)
        cv2.imshow('title', img)
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()
    
# Releasing the resource    
camera.StopGrabbing()

cv2.destroyAllWindows()

Bonitodelcapo, 2024-01-16:
> Hi! Yes, try with the following. You just need to fill the information for Destination Port and IP address and make sure you have those declared in the docker files. Let me know if that worked.
>
> ``` [...]

Hi Ignacio, could you also show how do you then map this info in your dockerfile?

Bonitodelcapo, 2024-01-17:
`           ip_address = 'XXX.XXX.XXX'
            tlf = pylon.TlFactory.GetInstance()
            cam_info = pylon.DeviceInfo()
            cam_info.SetIpAddress(ip_address)
            self.camera = pylon.InstantCamera(tlf.CreateDevice(cam_info))
            cam_info = self.camera.GetDeviceInfo()
            print(
                f"Name: {cam_info.GetModelName()}",
                f"IP: {cam_info.GetIpAddress()}",
                f"Mac: {cam_info.GetMacAddress()}"
            )`
            
 There is an error with specifically Assign "GigE". This snippet works for me

seb-koch, 2024-02-29:
+1, we're having the same issue. Using `host` mode isn't viable as it imposes a major security risk.

goncalog, 2024-03-22:
did you find an alternative @seb-koch?