basler/pypylon #124: Accessing camera by ip

Scoooooba, 2019-06-17:
Hey,
i have some problems connecting to a camera by ip address. By now i tried it with the following code

`ip_address = '172.20.2.1'
factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = ptl.CreateDeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = factory.CreateDevice(empty_camera_info)
cam_obj = pylon.InstantCamera(camera_device)`

which gives me the following error:

`Traceback (most recent call last):

  File "<ipython-input-92-e0a4c86c40fb>", line 6, in <module>
    camera_device = factory.CreateDevice(empty_camera_info)

  File ".......\site-packages\pypylon\pylon.py", line 1342, in CreateDevice
    return _pylon.TlFactory_CreateDevice(self, *args)

RuntimeException: Failed to discover GigE device '172.20.2.1:3956' : RuntimeException thrown (file 'baslergigetl.cpp', line 609)`

I also tried some other code 

`ip_address = '172.20.2.2'
info = pylon.DeviceInfo()
info.SetPropertyValue('IpAddress', ip_address)
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice(info))`

as suggested [here](https://github.com/basler/pypylon/issues/71), which seems to work fine until i try to change the exposure time:

`camera.ExposureTimeRaw.SetValue(5000)
Traceback (most recent call last):

  File "<ipython-input-94-252903592820>", line 1, in <module>
    camera.ExposureTimeRaw.SetValue(5000)

  File "........\site-packages\pypylon\genicam.py", line 2289, in SetValue
    return _genicam.IInteger_SetValue(self, Value, Verify)

AccessException: Node is not writable. : AccessException thrown in node 'ExposureTimeRaw' while calling 'ExposureTimeRaw.SetValue()' (file 'integert.h', line 77)`

I´d appreciate your help.

Best regards,
Scoooooba

thiesmoeller, 2019-06-17:
Call `cam_obj.Open()` before accessing the features...

The samples do it the right way. The readme.md sample has no parameter access,
So no need to open the control channel...

We definitely have to modify this example ;-)

Scoooooba, 2019-06-20:
Ah, easy, thank you!

SMA2016a, 2020-06-09:
info.SetPropertyValue("IpAddress","xxx.xxx.xxx.xxx") should work
or
info.SetIpAddress("xxx.xxx.xxx.xxx")

rdrandom, 2021-07-15:
Hello,
First of all, I'm sorry in advance for my English, it's not my mother tongue!
I am trying to use pylon to retrieve and save images taken from a camera present on the network and having a given ip address. I'm sorry if this sounds silly or if this problem has already occurred, I'm totally new to python. So I decided to follow this post to better understand and get there.
I tried, without success, to apply the answer given in comment. Could someone help me or explain it to me more precisely?
I thank you in advance !

closed 2024-02-28 by SMA2016a