basler/pypylon #125: The device cannot be operated on an USB 2.0 port.

keven425, 2019-06-19:
I'm using this USB3 camera: `Basler acA2440-75uc`. OS is ubuntu 18.04. Every now and then, I get this error. The error goes away, as soon as I unplug and plug back in.

Any idea? 

```
    device_tl = transport_layer.CreateDevice(device)
  File "/home/xkeven/.local/share/virtualenvs/unitx-VYNhvb4j/lib/python3.6/site-packages/pypylon/pylon.py", line 1342, in CreateDevice
    return _pylon.TlFactory_CreateDevice(self, *args)
_genicam.RuntimeException: Failed to open device '2676:ba02:1:6:7' for XML file download. Error: 'The device cannot be operated on an USB 2.0 port. The device requires an USB 3.0 compatible port.' : RuntimeException thrown (file 'PylonUsbTL.cpp', line 474)
```

thiesmoeller, 2019-06-19:
Typical sign of a worn-out connector on the host side, or a bad cable.

If the host detects that a stable superspeed connection is not possible, it falls back to slow usb2. Which is not supported on your model.

For reliable industrial USB 3 cables see: [Basler Cables](https://www.baslerweb.com/de/produkte/vision-komponenten/kabel/#series=baslerace;interface=USB30)

closed 2019-06-20 by keven425

Arthurgoujon, 2024-03-27:
I am having this exact issue. We have 6 different Basler Cameras and we are working with 6 different Nvidia Jetson AGX computers. The issue happens regularly and randomly on all those devices. This cannot be a hardware issue as all computers, cables and cameras are just out of the box. (Cables are balser's).

SMA2016a, 2024-03-28:
@Arthurgoujon  will you open a new thread,please?

ejlouw, 2025-04-14:
Same issue. Seems to occur randomly. Plugging the cameras out and in again seems to solve the problem (temporarily at least). Quite a big problem...

thiesmoeller, 2025-04-14:
This is related to electrical problems.
Do you have a worn out connector on your host,
Or in Nvidia case a custom carrier board?

You run into this error when the camera and the host controller are not able to create a stable superspeed link.