basler/pypylon #126: Hardware trigger questions

keven425, 2019-06-20:
1. Is there a configuration for hardware trigger, as `pylon.SoftwareTriggerConfiguration()`?

I've been using the following:

```
        camera.Open()
        camera.TriggerSelector.SetValue("FrameBurstStart")
        camera.TriggerSource.SetValue("Line1")
        camera.TriggerMode.SetValue("On")
```

2. Is there a way to `WaitForFrameTriggerReady()` for hardware trigger? When I call this before hardware trigger, I get a timeout error.

keven425, 2019-06-21:
Inspired by [this](https://github.com/basler/pypylon/issues/12#issuecomment-410861189), here's a working example. This allows hardware trigger with handler.

```
# define handler
# ...

# set up
camera.RegisterConfiguration(pylon.ConfigurationEventHandler(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)
camera.RegisterImageEventHandler(handler, pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
camera.Open()
camera.AcquisitionStatusSelector = "FrameTriggerWait"
camera.TriggerSelector.SetValue("FrameStart")
camera.TriggerMode.SetValue("On")
camera.TriggerSource.SetValue("Line1")
camera.TriggerActivation.SetValue('RisingEdge')
camera = self.init_image(camera)
camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)

# trigger
while True:
    camera.WaitForFrameTriggerReady(1000)
    # hardware trigger
```

closed 2019-06-21 by keven425

RaffaToSpace, 2019-07-24:
Hi, 
I am getting an error if using your example, it says the image type is not valid:

Cannot save image. The passed image is invalid. : InvalidArgumentException thrown (file 'imagepersistance.cpp', line 265)

whereas if I remove the last line 

camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)

I get a time out error.