basler/pypylon #127: macOS  whl file

FrankJin-SJ, 2019-06-21:
Binary release page shows " Added macOS platform support", which whl file is for MacOS system?  Thanks.

Resonanz, 2019-08-01:
Ditto, so frustrating that there is mention of it being supported, but no further info. Sigh.

thiesmoeller, 2019-08-02:
@Resonanz: you can compile it from the sources on your macOS system

nomel, 2019-09-14:
I was able to get it built on Mojave. Here's the instructions:

Download the source from: https://github.com/Basler/pypylon/releases

Extract and cd to the folder. Run: python setup.py bdist_wheel

You'll notice an error about "Unknown Platform" error in GCUtilities.h, around line 80. Open 
this file and add

```c
#define __LP64__ 1
```

around line 61, so it looks like this:

```c
#define USE_TEMP_CACHE_FILE 1
#endif // defined(UNDER_RTSS)

#define __LP64__ 1

#if defined (_MSC_VER)
#   if defined (_WIN64)
#       define PLATFORM_NAME "Win64_x64"
#   else
```

Try again and it should work. I don't know why __L64__ isn't being set. I believe it should be!

Btw, my GCUtilities.h could be found in: /Library/Frameworks/pylon.framework/Versions/A/Headers/GenICam/Base/GCUtilities.h

Yours may be in a different location. You can see where it is in the error message.

thiesmoeller, 2021-02-23:
Beginning with release 1.7.1 we have binary releases on all platforms, that you can install via pip.

update your pip version before by calling `python -m pip install --upgrade pip`

closed 2021-02-23 by thiesmoeller