basler/pypylon #128: USB3 camera handler callback unreliable

keven425, 2019-06-26:
We are running into an issue, where the number of times triggered vs number of images received by handler does not match.

We are using the separate thread for handling captures, to maximize fps:
https://github.com/basler/pypylon/blob/master/samples/grabusinggrabloopthread.py

We are currently using USB3 camera (Basler acA2440-75uc). We use a separate ImageEventHandler thread to handle the results. We are seeing this issue: if we trigger with maximum frame rate (with WaitForFrameTriggerReady(), so we are not over-triggering), sometimes handler receives fewer images than triggered. 

1. This issue only happens when more than one camera is used. It does NOT happen when only one camera is used.
2. This issue only happens when triggered at maximum frame rate. It does NOT happen when we add delays between triggers (~10 ms or above).

We are using ubuntu 18.04 kernal 4.18.0-21-generic. 

Any idea?

keven425, 2019-06-26:
Update: this issue only happens when there are more than one threads capturing at the same time. As soon as we add a lock around the capture method, everything works fine.

Question is: whether the driver not thread safe? I've reached out to basler tech support.

thiesmoeller, 2019-07-17:
see patch in
https://github.com/basler/pypylon/issues/116#issuecomment-499998088

keven425, 2019-07-17:
@thiesmoeller thanks I will try this out. Is there any plan to merge into master?

closed 2019-07-23 by keven425