basler/pypylon #129: How to set values for IPD and PacketSize by code?

lcflorindo, 2019-06-27:
Hi, everytime that I turn my whole system off, I must set the Packet Size and Inter-Packet Delay values throught pylon software before use the camera again (else I get the buffer error).

There is a way to do it by code? (Then everytime I run my programm it sets automatically)

Thanks in advance.

thiesmoeller, 2019-07-07:
You can always lookup the parameters in the normal camera documentation

[docs.baslerweb.com](https://docs.baslerweb.com/#t=en%2Fnetwork_related_parameters.htm)

The translation between C++ and python is nothing more than  _search-and-replace_:

Take care to call `camera.Open()` before accessing camera features.

### C++
```
// Payload Size
int64_t payloadSize = camera.PayloadSize.GetValue();
// Packet Size
camera.GevSCPSPacketSize.SetValue(1500);
// Inter-Packet Delay
camera.GevSCPD.SetValue(1000);
// Frame transmission delay
camera.GevSCFTD.SetValue(1000);
// Bandwidth Assigned
int64_t bandwidthAssigned = camera.GevSCBWA.GetValue();
// Bandwidth Reserve
camera.GevSCBWR.SetValue(10);
// Bandwidth Reserve Accumulation
camera.GevSCBWRA.SetValue(10);
// Frame Jitter Max
int64_t jitterMax = camera.GevSCFJM.GetValue();
// Device Max Throughput
int64_t maxThroughput = camera.GevSCDMT.GetValue();
// Device Current Throughput
int64_t currentThroughput = camera.GevSCDCT.GetValue();
// Heartbeat Timeout
camera.GevHeartbeatTimeout.SetValue(5000);
```

becomes:

### Python
```
# Payload Size
int64_t payloadSize = camera.PayloadSize.GetValue()
# Packet Size
camera.GevSCPSPacketSize.SetValue(1500)
# Inter-Packet Delay
camera.GevSCPD.SetValue(1000)
# Frame transmission delay
camera.GevSCFTD.SetValue(1000)
# Bandwidth Assigned
int64_t bandwidthAssigned = camera.GevSCBWA.GetValue()
# Bandwidth Reserve
camera.GevSCBWR.SetValue(10)
# Bandwidth Reserve Accumulation
camera.GevSCBWRA.SetValue(10)
# Frame Jitter Max
int64_t jitterMax = camera.GevSCFJM.GetValue()
# Device Max Throughput
int64_t maxThroughput = camera.GevSCDMT.GetValue()
# Device Current Throughput
int64_t currentThroughput = camera.GevSCDCT.GetValue()
# Heartbeat Timeout
camera.GevHeartbeatTimeout.SetValue(5000)

```

cute-tiger, 2019-11-22:
## Issue has been solved by adding these two lines into code:

#### Packet Size
camera.GevSCPSPacketSize.SetValue(1500) # using 9000 causes grabbing image is not successful every single time
#### Inter-Packet Delay
camera.GevSCPD.SetValue(2000)

lcflorindo, 2019-12-17:
Thanks for the support! It worked!

closed 2019-12-17 by lcflorindo

kesaroid, 2023-02-14:
I am using a acA1300-200uc, and when I try to set its attributes, I get the following error:
`Exception Node not existing (file 'genicam_wrap.cpp', line 16130)
Exception initializing camera`

This only happens when I try to set PacketSize and IPD. Acquisiting rate and exposure time methods don't seem to have the same problem

thiesmoeller, 2023-02-14:
@kesaroid .... You own a USB camera ... GIGE specific parameters are not available on your device.

Please check the docs.baslerweb.com for your camera specific parameters