basler/pypylon #130: Getting duplicate frame

HackersSpirit, 2019-06-28:
I using following model of basler camera: acA1920-40uc
I have connect my camera to controller which moves in a particular direction and capture images.
If i move camera from A to B to C. It happens that i get similar image for all the location.
I have tried **camera->StartGrabbing(GrabStrategy_LatestImageOnly)**; but i still get duplicate frame. I have used similar code in python and issue does not occur.
I can use grabOne and software trigger but it creates an extra delay to fetch the frame which is not desirable in my case. 
Sharing the cpp and python code:

Cpp:
Initialization:
```
void PrimaryCamera::initializeCamera()
{
    try
    {
        camera = new Pylon::CInstantCamera(CTlFactory::GetInstance().CreateFirstDevice());
        cout<<"Using Device"<<camera->GetDeviceInfo().GetModelName()<<endl;
        camera->MaxNumBuffer = 1;
        formatConverter.OutputPixelFormat = PixelType_BGR8packed;
        camera->Open();
        camera->StartGrabbing(GrabStrategy_LatestImageOnly);
        cout<<"\nCamera Initialized Successfully"<<endl;
    }catch(GenICam::GenericException &e) {
        cerr << "An exception occurred." << endl  << e.GetDescription() << endl;

    }

}
```
Capture Frame:
```
cv::Mat PrimaryCamera::captureBaslerFrame()
{
    if(camera->IsGrabbing()){
        camera->RetrieveResult(500, ptrGrabResult, TimeoutHandling_ThrowException);
        if (ptrGrabResult->GrabSucceeded()){
            formatConverter.Convert(pylonImage,ptrGrabResult);
            return cv::Mat(ptrGrabResult->GetHeight(),ptrGrabResult->GetWidth(), CV_8UC3,(uint8_t *)pylonImage.GetBuffer());

        }
    }else{
        cout<<"Currently camera is not grabbing frames"<<endl;
    }
}

```
Python:

thiesmoeller, 2019-06-28:
* the pylon APIs you use can't generate duplicate data, as there is a strict buffer management.
* As I don't see any trigger configuration, how to you synchronize the motion with the moments you want to capture an image.
* Without trigger the camera is free-running at 40fps with your model. So each frame you get is just taken  1/40 s apart from the frame before

HackersSpirit, 2019-07-01:
@thiesmoeller : Thanks for the response.

- Actually i haven't used software or hardware trigger because the time taken by s/w trigger gives  me frame in 100 to 120 millisec delay which is not desirable in my case and i haven't tried the hardware trigger.

- Secondly i for synchronization i move my controller by some distance give some time dealy for controller to stablize and then grab the frame.
 
- Following the same step in python give me distint image at each where as in cpp i get the same image/
If you like i will share the output of both  cpp and python.

-Please find the code below for clarity

-Python:
```
for i in range(3):
    #Writing data to serial port to move the controller by given distace
    ser.write(b"G0 X200 \n")
    ser.read()
    time.sleep(.2)
    img = grab_img()
    cv2.imwrite("path to save the image"+str(i)+".jpeg",img
```
Cpp 
```
for(int i =0; i<3; i++)
     {
       //Writing serial data to the controller
         serialPort->writeDataWithAck("G0 X200");
         QTest::qSleep(200);
         Mat sample = getBaslerFrame();
         imwrite("path to save image"+to_string(i)+".jpeg",sample);
       }
```
Please let me know in case of any issue.

HackersSpirit, 2019-07-03:
@thiesmoller @basler: one observation which i would like to share. When i run the above getbasler frame in while loop and continuously fetch the frames duplicate image cannot be seen seen. May be a sort of workaround but exactly haven’t figured the core issue.

thiesmoeller, 2019-07-03:
If you don't won't to synchronize with external hw trigger, but no the exakt timing
Of your movement use
https://docs.baslerweb.com/#t=en%2Facquisition_frame_rate.htm

HackersSpirit, 2019-07-04:
@thiesmoeller : Thanks again for your response.
Actually i wanted if there is problem in my cpp implemetation from the logic shared above?
This is not common issue shared by others and i tried to implement the same logic in my python code and the issue did not occur.
What my observation were when i continuously fetch in while loop the issue does not occur but when i go to a particular location and use the follwoing 
**camera->RetrieveResult(500, ptrGrabResult, TimeoutHandling_ThrowException);**
I get previous or duplicate frame.
I have used software trigger for the purpose but it takes more time to fetch the frame.

HackersSpirit, 2019-07-04:
@thiesmoeller : Thanks for the help. Finally i was able to find out the issue.
Actually earlier i was running a git example inorder to fetch the frame. When i went through the document i found out the problem was in initialization:
**camera->MaxNumBuffer = 1;**
The max size of the buffer the more samples can be stored.
This also answers the question why i was getting no repeated frame after continuously fetching the frame.

closed 2019-07-04 by HackersSpirit