basler/pypylon #131: How to set AutoExposure mode?

OlivierX, 2019-06-30:
I can use AutoExposure Mode on Basler Camera with the software Pylon Viewer, however, I'd use pypylon API in this repo to capture frame. But there is no information about parameters' setting in Nodemap.pfs. 
I've notice that there are relation settings like(Exposure Mode) in the file(Nodemap.pfs) and I've tried to set it to "Auto", but it returns error**"Feature 'ExposureMode' : cannot convert value 'Auto',"**

thiesmoeller, 2019-06-30:
Slight differences depending on you model.

You can use the documentation of the C++ API 

https://docs.baslerweb.com/#t=en%2Fexposure_auto.htm

OlivierX, 2019-07-01:
Thanks for your reply. Well,I supposed that Exposure Model in Nodemap.pfs is an option, and it would be more helpful if python API can work : )

thiesmoeller, 2019-07-01:
What software element do you prefer by 'Nodemap.pfs'?
pypylon has a Nodemap class. 

For further hints, due to 'Slight differences depending on you model'
What is your model

OlivierX, 2019-07-01:
Thanks, Moeller. 

Given by the sample code in this repositories(https://github.com/basler/pypylon/blob/c6f5a54fc7abb8c868ebfb37f8053ab69afd7270/samples/parametrizecameraloadandsaveconfig.py), it will generate an parametrized camera config file which is named by "Nodemap.pfs", and there is an option which is called "ExposureMode" which is set to "Timed" by default, and i couldn't figure out the  name of autoexposure modeafter several attemptations(Ex, Auto, Continuous).

And my model is aca2040-55um. So is it possible to avoid using C++ API by changing the parameters in the config file and loading it?

thiesmoeller, 2019-07-01:
The names in the pfs file are the names in the c++ API.
So just use the manual page that I referenced.
If you want to go this route:
Use pylon viewer to configure the camera
 exactly as you want and use 'save features' from.the menu to create the pfs file.

OlivierX, 2019-07-26:
Thanks moeller, it's been solved!

closed 2019-07-26 by OlivierX