basler/pypylon #132: Frames of GigE camera are inaccessible

matt4lieb, 2019-07-01:
**Setup:**
- Virtual machine (KVM) with ubuntu14.04, ubuntu16.04, ubuntu 18.04 and debian 9
- NIC is bridged, so every virtual machine has a own IP
- Only one virtual machine is running 
- Camera: GigE acA1920 - 25gc
- pylon 5.0.12 and 5.2

**Issue:**
I am able to connect and configure the camera on every linux OS, but my script only receives images on ubuntu 14.04.
Through wireshark I found out, that the camera is sending the images back to the system with the IP 0.0.0.0. Unfortunate it looks like, every OS above 14.04 is dropping the packages. My concern is, that newer kernel versions are dropping UDP packages, which don't have an IP. 

Is it possible to set the destination IP, so it is sending to a valid IP and not to 0.0.0.0? I tried it with the standard genicam parameter, but it didn't worked.

```
identifier = str(self.configuration["identifier"])
factory = pylon.TlFactory.GetInstance()

ptl = factory.CreateTl("BaslerGigE")
device_info = ptl.CreateDeviceInfo()
device_info.SetPropertyValue("IpAddress", camera_ip)
camera_device = factory.CreateFirstDevice(device_info)
camera = pylon.InstantCamera(camera_device)
camera.Open()
camera.GetNodeMap().GetNode("GevSCDA").SetValue(int(ipaddress.ip_address("host_ip")))
```
I always get a **pylon.TimeoutHandling_ThrowException**. Is there some "hack" to solve this? Because pylonviewer is working in the VMs. Be aware, that I will not able to connect via serial number. 

Thank you.

thiesmoeller, 2019-07-01:
Pylon viewer works ? Including streaming video?
Is the camera discovered directly , or do you have 
to give the device IP by hand?
Is the value in feature GevSCDA in the working case the same as in your sample code?

You can write more compact code;

`camera.GevSCDA = <intvalue>`

matt4lieb, 2019-07-01:
Pylonviever is working and the camera and host are internal ips. Discovery is not possible because of network restrictions, so serial number is not possible. 

The value which is given to the GevSCDA is the int based IP of the host. In the example I just replaced the IP with a placeholders.

thiesmoeller, 2019-07-01:
in your virtual machine, you can do simple wireshark capture when you run the pylonviewer.

And apply the following display filter:
`gvcp.cmd.writereg.bootstrapregister == 0x00000d18`

This will show the command, that sets the destination IP into the camera.

This should be the same, as in your programmatic use-case

thiesmoeller, 2019-07-01:
[image]

matt4lieb, 2019-07-02:
It probably does something to have with the StartGrabbing method. Because, when I am setting the destination ip it sends the request out to the GigE camera and when I call the StartGrabbing function it overwrites it. Do I have to do something different then `self.camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)` ? 
After setting the destination IP with `self.camera.GetNodeMap().GetNode("GevSCDA").SetValue(int(ipaddress.ip_address("10.1.11.251"))`
[image]
After StartGrabbing
[image]

The both commands are in the same order in the script, no other code is executed between them.

matt4lieb, 2019-07-02:
I also tried it with the pylon/Grab example and it is still the same issue with the Destination Address: 0.0.0.0
```
// Only look for GigE cameras.
CDeviceInfo info;
info.SetDeviceClass( BaslerGigEDeviceClass );
info.SetPropertyValue("IpAddress", "10.1.14.82");
// Create an instant camera object with the camera device found first.
CInstantCamera camera( CTlFactory::GetInstance().CreateFirstDevice(info));
        

// Print the model name of the camera.
cout << "Using device " << camera.GetDeviceInfo().GetModelName() << endl;
....

```

thiesmoeller, 2019-07-02:
What is the Wireshark trace in the working pylon viewer case?

Regarding your code: pylon sets SCDA internally on startgrabbing.

matt4lieb, 2019-07-03:
It looks good, with the correct IP. Is maybe the library doing something with sys-calls in the background, which can be blocked or in another format caused by the VM?

matt4lieb, 2019-08-12:
Is there are python function to call this method, like in c++?

`bool ret = gigeTL->AnnounceRemoteDevice("x.x.x.x", &info);`

kobotschick, 2020-07-25:
Do you found a solution to run pylon a VM?

IvanVishev, 2021-12-15:
Hello all!
I have already mentioned this problem in another issue about the 0.0.0.0 problem. 
I have almost the same configuration as OP and have the same problems. Does anyone have a solution?

IvanVishev, 2021-12-16:
I have investigated furhter on this problem. The same error with dest ip occurs also in pylon viewer. I am not sure,  but i think its a problem of ubuntu working on VM. Same camera works perfectly if accessed from Windows VM.

mlix11, 2021-12-21:
If you are using a qemu VM + NAT, you have to set the correct SNAT (e.g. iptables). Otherwise, it will not be able to find the camera.

closed 2021-12-21 by mlix11