basler/pypylon #134: Out of Memory when grabbing.

poisonous-milk, 2019-07-02:
I wrote a script to record videos and upload to cloud. 
After running for a couple minutes, I always end up with the following error. 

Traceback (most recent call last):
  File ".\stream_stable.py", line 139, in <module>
    image = converter.Convert(grab)
  File "C:\Users\Landing\Miniconda3\lib\site-packages\pypylon\pylon.py", line 6725, in Convert
    return _pylon.ImageFormatConverter_Convert(self, *args)
_genicam.BadAllocException: Out of memory. bad allocation : BadAllocException thrown (file 'bufferdata.h', line 52)

I am using LatestImageOnly strategy for grabbing, and I do release the grab result. Is this because of my processing is too slow?

thiesmoeller, 2019-07-05:
No. What you see is an out of memory error.
Can you break down your program into a small script that also aborts and post it?

Megaptera666, 2021-09-28:
I am getting the same error and cannot get around it. I believe the problem has to do with the converter but I dont know what is wrong. I am running a Basler acA1920 from a raspberry pi. I am setting the camera up as described [here](https://github.com/basler/pypylon/blob/master/samples/opencv.py) :

    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    camera.Open()
    print("Using camera: ", camera.GetDeviceInfo().GetModelName())

    camera.AcquisitionFrameRateEnable.SetValue(True)
    camera.AcquisitionFrameRate.SetValue(camera_frame_rate)
    camera.Width.SetValue(1920)
    camera.Height.SetValue(1088)

    # Set up converter for cv2 images to be able to store video
    converter = pylon.ImageFormatConverter()
    converter.OutputPixelFormat = pylon.PixelType_BGR8packed
    converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

I can run a function to generate a live preview and it works without problem. The code for the live preview is:

def camera_preview(camera, switch):
    """

    :param camera:
    :param switch:
    :return:
    """
    # conecting to the first available camera
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

    # Grabing Continusely (video) with minimal delay
    camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
    converter = pylon.ImageFormatConverter()

    # converting to opencv bgr format
    converter.OutputPixelFormat = pylon.PixelType_BGR8packed
    converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

    while camera.IsGrabbing():
        grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

        if grabResult.GrabSucceeded():
            # Access the image data
            image = converter.Convert(grabResult)
            img = image.GetArray()
            cv2.namedWindow('title', cv2.WINDOW_AUTOSIZE)
            cv2.imshow('title', img)
            k = cv2.waitKey(1)
            if k & switch.is_pressed:
                #camera.StopGrabbing()
                break
        grabResult.Release()

    # Releasing the resource
    camera.StopGrabbing()
    camera.Close()

    cv2.destroyAllWindows()

    return

However, when I try to incorporate this idea into a more complex script (the aim is to record video while I control some LED's with a predefined protocol), the script runs for several tens of seconds and then crashes. The section of the code that controls the camera is as follows:

            # time_before = dt.datetime.now()  # .strftime('%M_%S.%f')
            while camera.IsGrabbing():
                grabResult = camera.RetrieveResult(100, pylon.TimeoutHandling_ThrowException)

                if grabResult.GrabSucceeded():
                    # img = grabResult.Array
                    # Retrieve image and convert it to RGB space
                    image = converter.Convert(grabResult)
                    img = image.GetArray()
                    image_array.append(img)
                    # time_now = dt.datet(ime.now()  # .strftime('%M_%S.%f')
                    # deltat = (time_now - time_before)  # .strftime('%M_%S.%f')[:-3]
                    # print("time: ", deltat)
                    # time_before = time_now
                    txt = dt.datetime.now().strftime('%d-%m %H:%M:%S')
                    cv2.putText(img, txt, (img.size[0] // 2, img.size[1] // 10), cv2.FONT_HERSHEY_SIMPLEX, 1,
                                (255, 0, 0), 2)
                else:
                    print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
                grabResult.Release()

and the error I am getting is the same one pasted in the post above:

File "PATH", line xxx, in Convert
return _pylon.ImageFormatConverter_Convert(self, *args)
_genicam.BadAllocException: Out of memory. bad allocation : BadAllocException thrown (file 'bufferdata.h', line 52)

Any help would be greatly appreciated!!!

Megaptera666, 2021-09-28:
Turns out it is still the buffer size. If I record images that are 1080x1080 the program records 751 images before it crashes (consistently). If I grab 1920x1080 images then it records 415 images. 

The problem seems to be associated with storing the np.arrays/images. if I comment out the 

image_array.append(img)

line inside the " if grabResult.GrabSucceeded(): " loop then the code runs to completion (but its useless because I cannot save the images. Any insight @thiesmoeller ? Any help is greatly appreciated.

thiesmoeller, 2021-09-28:
@Megaptera666 :  
your image array has a size of 2.5 GB in the moment of "crash"
Because the ImageFormat converter is called _before_ appending the next image to your array, it will be the point where you simply run out of memory.

What is your OS and system?

thiesmoeller, 2021-09-28:
One option to store large image sequences is to use HDF5 ( https://www.hdfgroup.org/solutions/hdf5/)
Using the python module "h5py" this is fully integrated in your numpy/opencv/pypylon/... stack

I would really recommend to switch to a yuv format ( if possible in the camera pixelformat  ) to reduce the amount of data that is stored.
You could convert from yuv to rgb during processing.

Example to write image data to a hdf5 storage:

```python
import numpy as np
import pypylon.pylon as py
import h5py
import os
import time

class HDF5Store(object):
    """
    Simple class to append value to a hdf5 file on disc (usefull for building keras datasets)

    Params:
        datapath: filepath of h5 file
        dataset: dataset name within the file
        shape: dataset shape (not counting main/batch axis)
        dtype: numpy dtype

    Usage:
        hdf5_store = HDF5Store('/tmp/hdf5_store.h5','X', shape=(20,20,3))
        x = np.random.random(hdf5_store.shape)
        hdf5_store.append(x)
        hdf5_store.append(x)

    From https://gist.github.com/wassname/a0a75f133831eed1113d052c67cf8633
    """

    def __init__(self, datapath, dataset, shape, dtype=np.float32, compression="gzip", chunk_len=1):
        self.datapath = datapath
        self.dataset = dataset
        self.shape = shape
        self.i = 0

        with h5py.File(self.datapath, mode='w') as h5f:
            self.dset = h5f.create_dataset(
                dataset,
                shape=(0,) + shape,
                maxshape=(None,) + shape,
                dtype=dtype,
                compression=compression,
                chunks=(chunk_len,) + shape)

    def append(self, values):
        with h5py.File(self.datapath, mode='a') as h5f:
            dset = h5f[self.dataset]
            dset.resize((self.i + 1,) + self.shape)
            dset[self.i] = [values]
            self.i += 1
            h5f.flush()

tlf = py.TlFactory.GetInstance()
cam = py.InstantCamera(tlf.CreateFirstDevice())
cam.Open()

# setup data storage
hdf5_store = HDF5Store('hdf5_store.h5', 'images', compression=None, shape=(cam.Height.Value, cam.Width.Value, 3))
converter = py.ImageFormatConverter()
converter.OutputPixelFormat = py.PixelType_BGR8packed
cam.StartGrabbing()

while cam.IsGrabbing():
    grabResult = cam.RetrieveResult(500, py.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # img = grabResult.Array
        # Retrieve image and convert it to RGB space
        t0 = time.perf_counter()
        image = converter.Convert(grabResult)
        img = image.GetArray()
        hdf5_store.append(img)
        delta_ms = (time.perf_counter()-t0)*1e3
        print(f"captured frame{grabResult.ImageNumber} processing time {delta_ms}ms" )
    else:
        print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
    grabResult.Release()

```

How to access the data using numpy slice syntax
 
```python
import h5py

with h5py.File("hdf5_store.h5", mode='r') as h5f:
    dset = h5f["images"]

    # output the rgb values of the pixel [0,0] in all images
    print(dset[:,0,0,:])

```

thiesmoeller, 2021-10-07:
Some options:

You are using SD card storage ... This is slow. If you have RPi4 with Emmc .. use this

Don't convert the image format before saving. Do this during reading after recording.

Use a ram disc ( there are many tutorials how to do this )
This will allow you to use big portions of your 8GB RAM as storage 

And then copy h5py file to SD card after your recording