basler/pypylon #137: munmap_chunk(): Invalid Pointer

sleach3, 2019-07-15:
I have written and am trying to use a system that captures images from multiple basler cameras using pypylon. My initial tests were using two cameras and a multithread system to manage running the cameras and then recording the data. I was having an issue similar to #17 but I ran the given patch and that fixed my issues, for two cameras. When I added two more cameras to my test, I run into an issue where my collection runs for a few minutes and then either seg-faults, or the more common error I receive "**munmap_chunk(): Invalid pointer**". I am unsure where to look as I believe this error to be a C/C++ error and not a Python Error.

thiesmoeller, 2019-07-15:
There is an issue with threading.
Fix is https://github.com/basler/pypylon/issues/116#issuecomment-499998088

sleach3, 2019-07-15:
I apologize. I referenced the wrong post in my comment, and have edited the post. I applied that patch and still result with the munmap_chunk error.

thiesmoeller, 2019-07-15:
Please try to run your script with the memory debugger
`PYTHONMALLOC=malloc_debug python3 sample_script.py`

sleach3, 2019-07-15:
I still got the `munmap_chunk(): invalid pointer
Aborted
`
error.

thiesmoeller, 2019-07-15:
That's good. So no GIL related threading issue.

Turn on gdb.
Could you post the backtrace, when your program aborts?

`gdb --args python3 Yourscript.py`
`run`
... Till abort
`bt`

To dig deeper into where in the pypylon extension the error pops up

sleach3, 2019-07-16:
The results of running gdb on my code gives the following:

`#0  0x00007ffff514815a in std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007fffa11884a6 in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so
#2  0x00007fffa11885b7 in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so
#3  0x00007fffa1187d17 in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so
#4  0x00007fffa1176c5e in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so
#5  0x00007fffa1178941 in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so
#6  0x00007fffa118bbfa in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so
#7  0x00007ffff77cc6db in start_thread (arg=0x7fff90aeb700) at pthread_create.c:463
#8  0x00007ffff7b0588f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
`

sleach3, 2019-07-16:
My previous post was a seg-fault. I ran it again and got the munmap_chunk() error and got this output:

`#0  0x00007ffff7a22e97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51

#1  0x00007ffff7a24801 in __GI_abort () at abort.c:79

#2  0x00007ffff7a6d897 in __libc_message (action=action@entry=do_abort, 
fmt=fmt@entry=0x7ffff7b9ab9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181

#3  0x00007ffff7a7490a in malloc_printerr (str=str@entry=0x7ffff7b9c7a8 "munmap_chunk(): invalid pointer") at malloc.c:5350

#4  0x00007ffff7a7becc in munmap_chunk (p=0x1de1fa8) at malloc.c:2846

#5  0x00007ffff7a7becc in __GI___libc_free (mem=0x1de1fb8) at malloc.c:3117

#6  0x00007fffa11883b3 in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so

#7  0x00007fffa1188653 in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so

#8  0x00007fffa1187d17 in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so

#9  0x00007fffa1176c5e in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so

#10 0x00007fffa1178941 in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so

#11 0x00007fffa118bbfa in  ()
    at /home/62L/.local/lib/python3.6/site-packages/pypylon/libuxapi-5.2.0.so

#12 0x00007ffff77cc6db in start_thread (arg=0x7fff3c9e9700) at pthread_create.c:463

#13 0x00007ffff7b0588f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95`

thiesmoeller, 2019-07-16:
by checking the core libraries .... your are really running out of memory.

> I am currently using Ubuntu 18.04 and Python 3.6.8.

What is the CPU platform you use? 32bit/64bit?
e.g. on a 32bit system with typically just 1.5GB virtual memory for the user process you can easily run out of memory by operating multiple 20MPix cameras in YUV  [ 40MB per frame ] if you set the number of grab-buffers to a higher value.

To get a better idea about what your program is doing, you could use valgrind/massif as a tool to capture memory allocation behaviour:

```
sudo apt-get install valgrind
sudo apt-get install massif-visualizer

```
`PYTHONMALLOC=malloc valgrind --tool=massif python3 Yourscript.py`

The PYTHONMALLOC=malloc will disable the python builtin allocators to be visible by valgrind.

sleach3, 2019-07-16:
I am using a Intel® Xeon(R) CPU E5-1650 v3 @ 3.50GHz × 12 process with 32 GB of RAM and a 64-bit OS.

When I try to run the script with Valgrind and massif. It runs so slow that the camera retrieve times out. My fps that I calculate says that it is running at .06 frames per seconds when running with these tools.

thiesmoeller, 2019-07-16:
@sleach3, when installing the full pylon sdk for linux you would get a README, that contains some more hints in optimizing the USB subsystem for higher load ( which it is in your case with multiple cameras )

U3V Devices
-----------
* Increasing Packet Size
  For faster USB transfers you should increase the packet size. You can do this
  by changing the "Stream Parameters" -> "Maximum Transfer Size" value from
  inside the pylon Viewer or by setting the corresponding value via the API.
  After increasing the package size you will likely run out of kernel space
  and see corresponding error messages on the console. The default value set
  by the kernel is 16 MB. To set the value (in this example to 1000 MB) you can
  execute as root:
    echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb
  This would assign a maximum of 1000 MB to the USB stack.

thiesmoeller, 2019-07-16:
> I am using a Intel® Xeon(R) CPU E5-1650 v3 @ 3.50GHz × 12 process with 32 GB of RAM and a 64-bit OS.
>
> When I try to run the script with Valgrind and massif. It runs so slow that the camera retrieve [...]

... sorry for the misleading and deleted post. the valgrind and massif tool tip was a mis-posted for another ticket ....

sleach3, 2019-07-16:
Changing the usbfs_memory_mb variable has fixed my issue. Thank you very much.

closed 2019-07-16 by sleach3