basler/pypylon #138: how to set white balance auto

Thyagutamizhan, 2019-07-17:


MattsProjects, 2019-09-18:
Have a look at issue #146. There is a good explanation of using the user documentation to find information on the feature, and then going from the C++ code snippet to pypylon (the api's are identical save for the language).

closed 2020-12-01 by thiesmoeller

wweertman, 2020-12-16:
I also am having this issue. I was unable to follow this issue and find a solution. The C++ code for auto white balance in the GUI does not seem to translate directly to python. So I imagine it is just a syntax issue..

sjiang95, 2022-12-08:
There are sample codes for C/C++, etc. [Balance White Auto | Basler](https://docs.baslerweb.com/balance-white-auto#__tabbed_2_1). But don't know the corresponding interface in python.

thiesmoeller, 2022-12-08:
The mapping from c++ to python for *feature* access is really search and replace by this rule https://github.com/basler/pypylon/issues/505#issuecomment-1150780559