basler/pypylon #139: How to capture live video and displayed in the browser

gustavoponce7, 2019-07-19:
I am trying to display the live video from the camera in my browser, some kind of video streaming

Right now I am using the following code but when I consume my service (through http request) the UI is not refreshing the view

`def get_video():
    #conecting to the first available camera
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    #Grabing Continusely (video) with minimal delay
    camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
    converter = pylon.ImageFormatConverter()
    converter.OutputPixelFormat = pylon.PixelType_Mono16
    converter.OutputBitAlignment = "MsbAligned"
    iter = 0
    while camera.IsGrabbing():
        grabResult = camera.RetrieveResult(50000, pylon.TimeoutHandling_ThrowException)
        if grabResult.GrabSucceeded():
            # Access the image data
            retval, buffer = cv2.imencode('.jpg', grabResult.Array)
            base = str(b64encode(buffer))
            response = 'data:;base64,{}'.format(base[2:-1])
            iter += 1
            print(response[:100], iter)
            yield response
        grabResult.Release()
    # Releasing the resource
    camera.StopGrabbing()
    camera.Close()`

So I would like to know what is the best way to achieve this behavior

I am using the following tech stack:
- Camera: daA2500-14um
- Python version:  3.6
- Pypylon version: 1.4.0
- ReactJS as client in my browser

Thanks

MattsProjects, 2019-09-18:
If it helps, I have done a similar application via C/C++ and GStreamer, where GStreamer encodes the image stream as H.264 and transmits it over the network to another host which decodes it. I think maybe something like the VLC plugin could also be used to pickup the stream and show it in a browser, but I haven't tried that yet.
If you like to see the project, it is here:
https://imaginghub.com/projects/154-pylon-and-gstreamer

alexzauritz, 2020-01-13:
Hi @gustavoponce7, i am trying to do similar stuff. I want to display the live video from the camera in an android app. So I have to send the real time video to a server too. Did you solve your problem or do you have any advice?

thiesmoeller, 2020-01-13:
I've shown how to communicate with ffmpeg in this ticket: https://github.com/basler/pypylon/issues/113#issuecomment-543774545

Haven't tested it, but it should be possible to configure ffmpeg as an rtp source:
https://trac.ffmpeg.org/wiki/StreamingGuide

gustavoponce7, 2020-01-13:
Hi @alexzauritz , I have been able to implement it using a "**MotionJPEG**" strategy with Flask, here some info about it.

- Flask(web application framework), **mime type response** --> mimetype='multipart/x-mixed-replace; boundary=frame'
- Basically should be a python service generating frames, **returning type** --> yield (b'--frame\r\n'
                          b'Content-Type: image/jpeg\r\n\r\n' + self.__last_frame_generated + b'\r\n')
- It only works in Chrome

I hope it can be helpful for you

kobotschick, 2020-07-25:
Imagezmq does the trick

nchauhan5, 2021-03-11:
@kobotschick - Could you please share, how you got this working with pypylon and imagezmq, I am assuming you also had something like flask in there for the endpoint handler, serving the live video?

SilverLearn, 2022-08-14:
Your problem will be solved with the following project
This project was done by myself
https://github.com/SilverLearn/Real-time-display-of-Basler-cameras-on-web-server-in-Flask