basler/pypylon #142: Pypylon and pylon mono8 image format

PolymaTh-EE98, 2019-07-24:
Is the mono8 image format 3 channels? I’m using a line scan camera (runner I believe) and the image comes back 3 channels. Is that how the genicam standard also handles mono? It is quite vague... if it supposed to be single channel can assist me in finding the correct settings?

thiesmoeller, 2019-07-25:
A brief look into your camera manual is the first step :-) 

Mono is normally single channel data.

PolymaTh-EE98, 2019-07-25:
Then I have a software or a user error. Even in the pylon viewer when I hover over a mono8 image I get RGB values in the lower right. I asked another long time user and he also said it is 3 channel for him. What setting do I change to have mono8 behave as expected?

thiesmoeller, 2019-08-19:
What is the Pixelformat that you configure?

thiesmoeller, 2020-12-01:
no feedback since 2019

closed 2020-12-01 by thiesmoeller