basler/pypylon #144: Node not existing error

DilaraAlb, 2019-07-30:
Hi,

I got titled error for below code snippet, if you can help me I will appreciate.

```

cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))

# Create and attach all Pylon Devices.
for i, cam in enumerate(cameras):
	cam.Attach(tlFactory.CreateDevice(devices[i]))

	ip = cam.GetDeviceInfo().GetIpAddress()

	cam.Open() 

	cam.GainAuto.SetValue("Off")
	cam.Gain.SetValue(1.0)

```
The error is thrown for the last line `cam.Gain.SetValue(1.0)`

nomel, 2019-09-14:
Gain isn't a valid property.

See `dir(cam)` for valid properties.

You're probably looking for `cam.GainRaw`.

thiesmoeller, 2020-12-01:
the exact name of the feature can differ on your camera model ( in some few places  )

closed 2020-12-01 by thiesmoeller