basler/pypylon #145: Change view port on the fly

keven425, 2019-07-30:
We want to change the viewport (OffsetX, OffsetY, Width, Height) on the fly, while capturing async with grab handler callback. Here's a pseudo-code describing our use case:

```
# handler for grab callbacks
class Image_Event_Handler(pylon.ImageEventHandler):
    def OnImageGrabbed(self, camera, grab_result):
        image = self.converter.Convert(grab_result).GetArray()
        grab_result.Release()
        return image

# set up async grab handler 
handler = Image_Event_Handler()
camera.RegisterImageEventHandler(handler, pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)

# capture 30 images 
for (int i = 0; i < 30; i++)
    camera.capture_async()
 
# change view port to something else
camera.AcquisitionStop.Execute()
camera.OffsetX = <new value>
camera.OffsetY = <new value>
camera.Width = <new value>
camera.Height = <new value>
camera.AcquisitionStart.Execute()
 
# capture 3 images
for (int i = 0; i < 3; i++)
    camera.capture_async()
```

On resizing the viewport, we run into the following error:

```
    camera.Width = width
  File "/home/xkeven/.local/share/virtualenvs/unitx-VYNhvb4j/lib/python3.6/site-packages/pypylon/pylon.py", line 3840, in __setattr__
    self.GetNodeMap().GetNode(attribute).SetValue(val)
  File "/home/xkeven/.local/share/virtualenvs/unitx-VYNhvb4j/lib/python3.6/site-packages/pypylon/genicam.py", line 2188, in SetValue
    return _genicam.IInteger_SetValue(self, Value, Verify)
_genicam.AccessException: Node is not writable. : AccessException thrown in node 'Width' while calling 'Width.SetValue()' (file 'IntegerT.h', line 77)
```

Any idea?

thiesmoeller, 2019-07-30:
Changing the viewport size changes the amount of data transferred.
As all DMA into the receive buffers  is set up already, the grab engine forbids the wires into Width and Height.

Is it required to change the size ?
... There are ways to change on the fly ... I can check if I find a sample for it ....

PolymaTh-EE98, 2019-07-30:
I would also like to change height on the fly or at least for the next frame...

keven425, 2019-07-30:
@thiesmoeller yes it would be ideal to change on the fly. We have two capture scenarios that prioritizes view-port size vs. frame rate.

thiesmoeller, 2019-07-31:
Tested on a Basler ace USB:

Output of the script below:

```
TI trigger_first  0 	W 1936  H 1216  X    0  Y    0  TS_TRG          0  ID        0  
TI trigger_first  1 	W 1936  H 1216  X    0  Y    0  TS_TRG  102899133  ID        1  
TI trigger_first  2 	W 1936  H 1216  X    0  Y    0  TS_TRG  205669854  ID        2  
TI trigger_first  3 	W 1936  H 1216  X    0  Y    0  TS_TRG  308543580  ID        3  
TI trigger_first  4 	W 1936  H 1216  X    0  Y    0  TS_TRG  411483024  ID        4  
TI trigger_first  5 	W 1936  H 1216  X    0  Y    0  TS_TRG  514564272  ID        5  
TI trigger_second  0 	W  484  H  304  X    0  Y    0  TS_TRG  538759107  ID        6  
TI trigger_second  1 	W  484  H  304  X    0  Y    0  TS_TRG  640270683  ID        7  
wait for completion
TI trigger_second  2 	W  484  H  304  X    0  Y    0  TS_TRG  742185396  ID        8  

```

```
from pypylon import pylon
from queue import Queue
import time

# sample application for viewport changes during active background grabbing

class AsyncImageDataHandler(pylon.ImageEventHandler):
    """
    Handler for the async recv of the video data
    """

    def __init__(self, trigger_ctrl):
        super().__init__()
        self.converter = pylon.ImageFormatConverter()

        # Second the converter must be parameterized.
        self.converter.OutputPixelFormat = pylon.PixelType_Mono8
        self.converter.OutputBitAlignment = "MsbAligned"
        self.trigger_ctrl = trigger_ctrl
        self.format_string = """TI {trigger_info:s} \tW {width:4d}  H {height:4d}  X {offsetx:4d}  Y {offsety:4d}  TS_TRG {ts_trg:10d}  ID {id:8d}  """
        self.first_ts = None

    def OnImageGrabbed(self, camera, grabResult):
        # best practice to install a generic exception handler to see errors in the callback background handler
        # during development
        try:
            # extract image info
            width = grabResult.GetWidth()
            height = grabResult.GetHeight()
            offsetx = grabResult.GetOffsetX()
            offsety = grabResult.GetOffsetY()
            timestamp = grabResult.GetTimeStamp()

            if self.first_ts is None:
                self.first_ts = timestamp
                self.first_ts = timestamp

            timestamp = (timestamp - self.first_ts)

            # counter of transmitted frame from camera
            block_id = grabResult.GetBlockID()

            # fetch the async trigger event data submitted by trigger call
            capture_info = self.trigger_ctrl.get_capture_info()

            print(self.format_string.format(width=width, height=height, offsetx=offsetx, offsety=offsety,
                                            ts_trg=timestamp, id=block_id, trigger_info=capture_info))

            # img = self.converter.Convert(grabResult).GetArray()
            # do something ... with image data

        except Exception as e:
            print("Exception in OnImageGrabbed: ", e)
            # false -> triggers a StopGrabbing call
            raise e

        # true tells the grab engine, that we want to move on receiving images
        return True

class TriggerController(object):
    def __init__(self, camera):
        self.camera = camera
        # queue of tuples of ( trigger_timestamp in camera, trigger_info )
        self._trigger_info = Queue()
        # set up the Acquisition Status to monitor the "ready for next trigger" signal
        self.camera.AcquisitionStatusSelector = "FrameTriggerWait"

    def capture_async(self, trigger_info):
        # wait for camera ready
        while not self.camera.AcquisitionStatus.Value:
            time.sleep(0.1)
        # trigger the camera
        self.camera.TriggerSoftware.Execute()

        self.__set_capture_info(trigger_info)

    def __set_capture_info(self, info_val):
        self._trigger_info.put(info_val)

    def get_capture_info(self):
        val = self._trigger_info.get()
        return val

    def has_unprocessed_buffers(self):
        buf_unproc = self._trigger_info.empty()
        return not buf_unproc

# initialize the objects
# Create an instant camera object for the camera device found first.
cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
# open device to get access to all features
cam.Open()

# reset to default config before all other settings
cam.UserSetSelector = "Default"
cam.UserSetLoad.Execute()

cam.ExposureTime = cam.ExposureTime.Min

controller = TriggerController(cam)

image_handler = AsyncImageDataHandler(controller)

cam.RegisterImageEventHandler(image_handler, pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)

cam.AcquisitionMode = "Continuous"
cam.TriggerMode = "On"
cam.TriggerSelector = "FrameStart"
cam.TriggerSource = "Software"

cam.MaxNumBuffer = 10

# to capture variable frame sizes with a single grab setup, the largest
# image size that is possible has to be configured before StartGrabbing
# this defines the receivebuffer sizes in the grab engine.

# e.g. toggle between full-ROI and top left quarter-AOI

cam.OffsetX = 0
cam.OffsetY = 0
cam.Width = cam.Width.Max
cam.Height = cam.Height.Max

cam.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)

for i in range(6):
    controller.capture_async("trigger_first %2d" % i)

# to set image roi ( not Offset!! ) during capture the engine has to be unlocked
# and the camera acquisition stopped/started
# you have to guarantee that new-ROI <= max-ROI

# cam stop
cam.AcquisitionStop.Execute()
# grab unlock
cam.TLParamsLocked = False

cam.OffsetX = 0
cam.OffsetY = 0
cam.Width = cam.Width.Max // 4
cam.Height = cam.Height.Max // 4

# grab lock
cam.TLParamsLocked = True
# cam start
cam.AcquisitionStart.Execute()

for i in range(3):
    controller.capture_async("trigger_second %2d" % i)

# wait for completion of all buffers
print("wait for completion")
while controller.has_unprocessed_buffers():
    time.sleep(.5)

cam.StopGrabbing()

```

closed 2020-12-01 by thiesmoeller