basler/pypylon #146: How can I access python API documentation for pylon? Or do you provide any?

DilaraAlb, 2019-07-31:


thiesmoeller, 2019-07-31:
There are multiple documentation sources:

The camera features are provided dynamically by the camera itself [ genicam interface ]
So you have to lookup the docmentation on https://docs.baslerweb.com

The syntax between C++ and python is nearly identical. So you can always use the C++ help snippets in the online-documentation

Example for Gain for an USB camera
[image]

Example for Gain for a GigE camera
[image]

The pypylon core, grabbing video and controlling the camera has integrated python docu:

`pydoc pypylon.pylon`

`pydoc pypylon.genicam`

although some of the docstrings do still reveal some of their C++ heritage ;-)

DilaraAlb, 2019-07-31:
Can you be more specific? I have clicked given url and could not find the Gain example. Of course, it exists but it is an extensive menu and couldn't find direct link to the api.

For the docs accessed with the commands, is there a way for me to save it as text/pdf? It is not user friendly as it is.

elishatam, 2019-09-16:
Here's a specific link to the Gain example: https://docs.baslerweb.com/index.htm#t=en%2Fgain.htm
Sample code is at the bottom of the page.

I also found some API information in a folder within Pylon Viewer, located here on the computer:
C:\Program Files\Basler\pylon 6\Development\Documentation\PylonCppSDK

Cody-G, 2020-03-12:
In a Debian install the C++ docs seem to end up here:

`file:///opt/pylon5/share/doc/C++/index.html`

I haven't found a pdf yet.  Definitely not user-friendly.

thiesmoeller, 2020-03-12:
Do you expect more of a "book-style" documentation?

The referenced index.html is the entry to the C++ API documentation on Linux.
As pypylon is a python binding to our C++ API, All concepts are the same. For camera features accessed by pypylon, you should check docs.baslerweb.com to get your model-specific details. 

[image]

Cody-G, 2020-03-12:
>Do you expect more of a "book-style" documentation?
>
>The referenced index.html is the entry to the C++ API documentation on Linux.
As pypylon is a python binding to our C++ API, All concepts are the same. For camera features accessed by pypylon, you should check docs.baslerweb.com to get your model-specific details.

Thanks for your prompt response!  The C++ documentation is relatively good, and it works well to document the Python API as well.  What I think is unfriendly to the user is that in order to _find_ the documentation I had to:

1. Install both the SDK and the Pylon viewer application
2. Open the Pylon viewer application and find the documentation link under the `?` menu entry.
3. Find the browser tab opened by the link (my browser didn't pop up after clicking the link).

Then, since the documentation is in a series of web pages instead of a pdf it's more difficult to search.  That's a relatively minor issue for me.  If I could make one request it's that the SDK documentation be hosted online somewhere.  (I realize that's not the responsibility of the pypylon repo, but just thought I'd say this while I have your attention).

thiesmoeller, 2020-03-13:
Thanks for the feedback.
We'll check better ways to link pypylon users to pylon c++ docu.
The HTML based documentation has a quite good integrated search. It is in the top right of the screen.

lvlanson, 2021-09-05:
is there currently a way to get the documentation without installing the software?

ethanc8, 2024-06-26:
Basler's C++ API documentation is viewable at https://docs.baslerweb.com/pylonapi/cpp/ now.

thiesmoeller, 2024-06-26:
We now have for every feature in the product documentation python c c++ and c# documentation