basler/pypylon #148: Unable to change TriggerMode

SpangeJ, 2019-08-09:
Hello,
I am configuring two Basler acA1920-155um in python.
An excerpt of my code is
`
for i in range(2):
`
 `           self.Camera[i].TriggerSource     = "Line3"`
 `           self.Camera[i].TriggerSelector   = "FrameBurstStart"`
 `           self.Camera[i].TriggerMode        = 'On'`
  `          self.Camera[i].LineSelector        = "Line3"`
   `         self.Camera[i].LineMode             = "Input"`

But when I try to trigger the camera, with an external signal, I get the `TimeoutHandling_ThrowException`
When I open pylon Viewer I see that TriggerMode is `Off` and LineSelector is `Line1`
If I change these, in pylon viewer, (to `On` and `Line3`, respectfully) and rerun the external trigger signal, it works.

Any thought why e.g. ` self.Camera[i].TriggerMode        = 'On'` don't crash, but don't set the property?

thiesmoeller, 2019-08-09:
what type is `self.Camera` ?

SpangeJ, 2019-08-09:
`print(type(self.Camera[i]))` gives:
`<class 'pypylon.pylon.InstantCamera'>`

thiesmoeller, 2019-08-09:
... in principle this should work .....  ;-)

do you call `RegisterConfiguration` in your code and after doing your config calls? 
So please read the documentation [ available on full pylon sdk install ]. They are exactly used to _set_ configurations of the devices from a central place. 
And will consequently overwrite all your manual settings.

If you want to use them, just inherit like in [configurationeventprinter.py](samples/configurationeventprinter.py)

```
# run with emulated devices
import os

os.environ["PYLON_CAMEMU"] = "3"
import pypylon.pylon as pylon

class CameraBasler(object):

    def __init__(self):

        device_class = "BaslerCamEmu"
        di = pylon.DeviceInfo()
        di.SetDeviceClass(device_class)

        tlFactory = pylon.TlFactory.GetInstance()

        # Get emu devices.
        # TODO: funny syntax because Enumerate Devices still requires a device info list .....
        devices = tlFactory.EnumerateDevices([di, ])

        self.camera = []

        # create camera devices
        for i, cam in enumerate(devices):
            self.camera.append(pylon.InstantCamera(tlFactory.CreateDevice(devices[i])))

    def open_cams(self):
        for i in range(2):
            self.camera[i].Open()

    def setup_cams(self):
        for i in range(2):
            print(self.camera[i].DeviceModelName.Value, type(self.camera[i]))
            self.camera[i].TriggerSource = "Software"
            self.camera[i].TriggerSelector = "AcquisitionStart"
            self.camera[i].TriggerMode = 'On'

    def show_status(self):
        for i in range(2):
            print(self.camera[i].TriggerSource.Value, end=",")
            print(self.camera[i].TriggerSelector.Value, end=",")
            print(self.camera[i].TriggerMode.Value)

cams = CameraBasler()
cams.open_cams()
cams.setup_cams()
cams.show_status()

```

gaeeronimo, 2019-08-12:
`TriggerSource` & `TriggerMode` are _selected_ by `TriggerSelector`, so you need to set `TriggerSelector` **first** to the Trigger you want to configure (eg. `FrameBurstStart` if you want to trigger a burst of frames). After _selecting_ the Trigger you may configure it (eg. enable it by `TriggerMode = 'On'` and set the source by `TriggerSource = 'Line3'`).

Try it this way:

```
for i in range(2):
    self.Camera[i].TriggerSelector = "FrameBurstStart" # <== Selector first!
    self.Camera[i].TriggerSource = "Line3"
    self.Camera[i].TriggerMode = 'On'

    self.Camera[i].LineSelector = "Line3"
    self.Camera[i].LineMode = "Input"
```

SpangeJ, 2019-08-12:
> `TriggerSource` & `TriggerMode` are _selected_ by `TriggerSelector`, so you need to set `TriggerSelector` **first** to the Trigger you want to configure (eg. `FrameBurstStart` if you want to trigger [...]

Thank you, @gaeeronimo, this was precisely it!
I changed the order, such that `TriggerSelector` was prior to `TriggerSource` & `TriggerMode`  and now
 `self.Result[i] = self.Camera[i].RetrieveResult(5000,pylon.TimeoutHandling_ThrowException)` 
runs smoothely!

closed 2019-08-12 by SpangeJ