basler/pypylon #149: Pylon Viewer on VMWare

keivanK1, 2019-08-22:
Hi,
I want to do "single/continuous shot" using Pylon Viewer in Debian 10 which is installed on VMWare. My host is windows 10.

On the host side everything works. I enabled "Jumbo Packet" for network interface in windows and it's value is 9014 Bytes. Under the host (Windows 10) I can stream from the camera "Basler acA1920-50gc" correctly if I change the packet size to 8192 in Pylon Viewer (as described in installation guide for pylon 5.2).
But I can't stream it under the virtual machine. Basically, I have set the MTU of it's network to 9014 and the packet size under transport layer is set to 8192, but it is not working and give me the following error:

> Type	Time	Source	Message
Error	2019-08-22 04:14:57.622	Basler acA1920-50gc (21967297)	Image acquisition on "Basler acA1920-50gc (21967297)" failed! Error: 0xe1000014 "The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size."

And the picture I get is black. If I set the Packet size to 1500, it will give me the picture but not completely, basically it contains dick black lines and after capturing the same error comes.

Is there something that I' missing?

thiesmoeller, 2019-08-23:
If you get no imagedata at all, this is a sign for camera configured for jumbo packets but VMware not ....

Have you increased the InterPacketDelay as the error message said?
The camera in default settings might send the packets faster ( with less spacing between them) as VMware can handle it ...

keivanK1, 2019-08-23:
If I change packet size = 8192 I will not get any image but by 1500 I will get. 
Yes, I've increased the InterPacketDelay and it works by packet size = 1500 but the delay is too much.
To me, it seems the problem is exactly the Jumbo packet in VMWare. Although it set to 9014, but in reality, it does not work.
Do you mean that it is not possible to stream the camera in VMWare?

closed 2019-10-22 by keivanK1