basler/pypylon #150: boost thread: trying joining itself: Resource deadlock avoided

keven425, 2019-08-30:
Hi, every now and then, I get the following error:

```
terminate called after throwing an instance of 'baslerboost::exception_detail::clone_impl<baslerboost::exception_detail::error_info_injector<baslerboost::thread_resource_error> >'
  what():  boost thread: trying joining itself: Resource deadlock avoided
```

I haven't figured out a way to reproduce it. I have 4 cameras running in two threads (two cameras per thread). 

Any idea?

thiesmoeller, 2019-08-31:
could you provide version of pypylon/python/os/camera-interface?

keven425, 2019-09-03:
@thiesmoeller where is this `pypylon/python/os/camera-interface` located? I checked the `site-packages/pypylon` directory, but didn't find it there.

Thanks.

thiesmoeller, 2019-09-04:
i meant: please report
* pypylon version
* python version
* os
* camera-interface

have you applied the patch in https://github.com/basler/pypylon/issues/128 ?

thiesmoeller, 2019-09-05:
Do you have an `ImageEventHandler`, where you call `StartGrabbing `in the method `OnImageGrabbed `[ which is driven by pylon internal thread]. This ends in the join you reported.

Possibly in some sort of error handling?

keven425, 2019-09-05:
@thiesmoeller yes I do have an `ImageEventHandler`. Code is attached below. I only call `StartGrabbing` in the main thread, not inside `OnImageGrabbed` though. 

Do you think the issue is caused by error inside `OnImageGrabbed`?

```
class Image_Event_Handler(pylon.ImageEventHandler):

    def OnImagesSkipped(self, camera, n_skipped_images):
        self.logger.error("image skipped event for device ", camera.GetDeviceInfo().GetModelName())
        self.logger.error(n_skipped_images, " images have been skipped.")
        raise CameraRuntimeException('images skipped')

    def OnImageGrabbed(self, camera, grab_result):
        self.logger.debug('%f %s grabbed' % (time.time(), self.sn))
        # Image grabbed successfully
        if not grab_result.GrabSucceeded():
            self.logger.error("retrieve timeout: ", str(grab_result.ErrorCode), grab_result.ErrorDescription)
            raise CameraTimeoutException('%s camera retrieve timeout exceeded' % self.sn)
        
        # handle image
        # ...
```

thiesmoeller, 2019-09-05:
If you return False from the image event handler an implicit stopgrabbing is called.
The exception is not transferrered to main thread.

keven425, 2019-09-05:
I see. So in my `OnImageGrabbed`, I either:

1. return nothing, or
2. raise exception

Would any of that cause the deadlock error? Is there a best practice? (e.g. always catching exception and return true)

thiesmoeller, 2019-09-18:
> always catching exception and return true

yes this is what i do.
decide in the exception handler if can proceed -> true
if this is really a stop -> false

and output the exception contents to a logger channel

closed 2019-09-18 by keven425

simonaertssen, 2020-08-07:
Hi, I'm having some difficulties with subclassing the pylon.ImageEventHandler, as I cannot find an implementation with a more elaborate example than in the sample grabusinggrabloopthread.py

I have made an implementation as found below, and the grabResult is saved as an attribute of the class, which I can access from outside the class, as I wish to process the result somewhere else. I would expect some sort of signal to be available when a new grabResult is registerd (eg when OnImageGrabbed is called). How can I make this more robust?

class ImageEventHandler (pylon.ImageEventHandler):

    def __init__(self, callback):
        super().__init__()
        self.returnImageCallback = callback
        self.grabbedImage = None

    def OnImageGrabbed(self, camera, grabResult):
        if grabResult.GrabSucceeded():
            with grabResult.GetArrayZeroCopy() as ZCArray:
                self.grabbedImage = ZCArray
            self.returnImageCallback(self.grabbedImage)