basler/pypylon #152: "Ambiguous device info properties" with CreateDevice and UserDefinedName

nomel, 2019-09-14:
When trying to create a device using a UserDefinedName, a RuntimeError occurs with  "Ambiguous device info properties".

To reproduce:
```python
from pypylon import pylon

# I have two GigE cameras. The user defined names are "UpCCD" and "DownCCD"
names = [device.GetUserDefinedName() for device in pylon.TlFactory.GetInstance().EnumerateDevices()]
# >>> ['DownCCD', 'UpCCD']

# Trying to use one of these names results in the error
device_info = pylon.DeviceInfo()
device_info.SetUserDefinedName(names[0])  # or device_info.SetPropertyValue('UserDefinedName', 'UpCCD')
device = pylon.TlFactory.GetInstance().CreateDevice(device_info)
```

```text
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/IPython/core/interactiveshell.py", line 2882, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-18-e753eeb0c038>", line 7, in <module>
    device = pylon.TlFactory.GetInstance().CreateDevice(device_info)
  File "/Library/Python/2.7/site-packages/pypylon/pylon.py", line 1060, in CreateDevice
    return _pylon.TlFactory_CreateDevice(self, *args)
RuntimeException: Ambiguous device info properties. 2 devices match the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 695)
```

MattsProjects, 2019-09-18:
Just to check, do you only see one instance of each camera in the Pylon Viewer?
It is possible to have duplicate enumerations if you have two cameras, a NIC for each one, and both NICs on the same subnet (ie: NIC 1 = 192.168.1.101/255.255.255.0, NIC 2 = 192.168.1.102/255.255.255.0). If that is the case, the NICs must run on different subnets, like 192.168.**1**.101 and 192.168.**2**.101, (both with 255.255.255.0 masks).

Unmoon, 2020-10-07:
I'm also seeing this issue - Pylon Viewer only shows one instance, and I'm only using one NIC.

```python
pylon.TlFactory.GetInstance().EnumerateDevices()
(<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x10ed4e180> >,
<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x1261a6750> >)
```

```python
tuple(numpy.unique(pylon.TlFactory.GetInstance().EnumerateDevices()))
(<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x1261a6cc0> >,)
```

closed 2024-07-19 by SMA2016a