basler/pypylon #153: Stitch 2 camera images into 1 GUI

elishatam, 2019-09-16:
I have 2 cameras and would like to place both images side-by-side into 1 GUI (Pylon Image Window). I've reviewed grabmultiplecameras.py and guiimagewindow.py as reference. Currently I can create a GUI for each camera and place the GUI's side-by-side.

How can I stitch 2 separate camera images together into 1 GUI?  
Thank you

MattsProjects, 2019-09-16:
Hi Elishatam,
I've done this before in C++ if that helps. You can find my library with a sample in the comments at the bottom here:
https://github.com/MattsProjects/PylonSample_Stereo_Acquisition/blob/master/PylonSample_Stereo_Acquisition/include/StitchImage.h
-Matt

elishatam, 2019-09-18:
Thanks Matt.
I figured out how to get something working by using OpenCV. I referenced 3 of the code samples and combined them together.
1) grabmultiplecameras.py
2) guiimagewindow.py
3) opencv.py

Here's my code
```
# Grab_MultipleCameras.cpp
# ============================================================================
# This sample illustrates how to grab and process images from multiple cameras
# using the CInstantCameraArray class. The CInstantCameraArray class represents
# an array of instant camera objects. It provides almost the same interface
# as the instant camera for grabbing.
# The main purpose of the CInstantCameraArray is to simplify waiting for images and
# camera events of multiple cameras in one thread. This is done by providing a single
# RetrieveResult method for all cameras in the array.
# Alternatively, the grabbing can be started using the internal grab loop threads
# of all cameras in the CInstantCameraArray. The grabbed images can then be processed by one or more
# image event handlers. Please note that this is not shown in this example.
# ============================================================================

import os

os.environ["PYLON_CAMEMU"] = "3"

from pypylon import genicam
from pypylon import pylon
import sys
import time
import cv2
import numpy as np
# Number of images to be grabbed.
countOfImagesToGrab = 10

# Limits the amount of cameras used for grabbing.
# It is important to manage the available bandwidth when grabbing with multiple cameras.
# This applies, for instance, if two GigE cameras are connected to the same network adapter via a switch.
# To manage the bandwidth, the GevSCPD interpacket delay parameter and the GevSCFTD transmission delay
# parameter can be set for each GigE camera device.
# The "Controlling Packet Transmission Timing with the Interpacket and Frame Transmission Delays on Basler GigE Vision Cameras"
# Application Notes (AW000649xx000)
# provide more information about this topic.
# The bandwidth used by a FireWire camera device can be limited by adjusting the packet size.
maxCamerasToUse = 2

# The exit code of the sample application.
exitCode = 0
img0 = []
img1 = []
windowName = 'title'

try:

    # Get the transport layer factory.
    tlFactory = pylon.TlFactory.GetInstance()

    # Get all attached devices and exit application if no device is found.
    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RUNTIME_EXCEPTION("No camera present.")

    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))
    
    l = cameras.GetSize()

    # Create and attach all Pylon Devices.
    for i, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[i]))
[... 66 lines cut ...]
                cv2.destroyAllWindows()
                break

        else:
            print("Error: ", grabResult.ErrorCode)
            # grabResult.ErrorDescription does not work properly in python could throw UnicodeDecodeError
        grabResult.Release()
        time.sleep(0.1)

        #If window has been closed using the X button, close program
        # getWindowProperty() returns -1 as soon as the window is closed
        if cv2.getWindowProperty(windowName, 0) < 0:
            cv2.destroyAllWindows()
            break
            

except genicam.GenericException as e:
    # Error handling
    print("An exception occurred.", e.GetDescription())
    exitCode = 1

# Comment the following two lines to disable waiting on exit.
sys.exit(exitCode)

```

closed 2019-09-18 by elishatam