basler/pypylon #154: Support for getting coreboard temperature

thedim-witted, 2019-09-19:
I couldn't find a python api for reading the coreboard temperature and was wondering if there is a way to do it.

thiesmoeller, 2019-09-19:
It's a feature of your camera, that will just be made available by pypylon.
So check on docs.baslerweb.com the name of the feature for your model ...

thedim-witted, 2019-09-20:
Yes, there is a feature but I need a python API to use it in my script. All I could find is C and C++ commands.

thiesmoeller, 2019-09-20:
As you don't mention your camera model, I can't give you the exact answer.....

But if the documentation states:
```
// Get the current device temperature
double d = camera.DeviceTemperature.GetValue();
```

in python it is just
```
# Get the current device temperature
d = camera.DeviceTemperature.GetValue()
```

All camera functions are always exactly as their C++ counterpart. because the API is coming from your camera itself.

thedim-witted, 2019-09-20:
I did the same by using the commands mentioned in the documentation. But it throws an error when I tried to initialize the temperature sensor. However, I guess I don't have to do that step in python and directly read the value.

closed 2019-09-20 by thedim-witted

thiesmoeller, 2019-09-20:
Please mention the camera model in issue tickets, so an exact answer is possible.

rohitrkr, 2022-05-20:
@thedim-witted did you find any solution for the same..??
not able to get device temperature by this line of code in python camera.DeviceTemperature.GetValue()
my devce model number is acA1920-40gc.
@thiesmoeller

ckldas, 2022-05-23:
@thedim-witted @thiesmoeller Pls help......

ckldas, 2022-05-23:
Not able to Read Basler Camera Core Board Temperature Via pypylon

ckldas, 2022-05-24:
yes

ckldas, 2022-05-24:
@thedim-witted u closed this issue. Pls let me know if u got the exact solution.

thiesmoeller, 2022-05-24:
@ckldas 
As you can read in pylon viewer, you can also read in pypylon

In the pylon viewer documentation panel there is always a C++ snippet, that you can copy nearly 1:1

As you never copied your error message it is hard to support

ckldas, 2022-05-24:
d = camera.DeviceTemperature.GetValue()

Error is "camera dont have attribute DeviceTemperature".

thiesmoeller, 2022-05-24:
as this is not an error message, that pypylon would throw ....

here is a short demo code to show the temperature sensors of a camera:
```python
import pypylon.pylon as py

tlf = py.TlFactory.GetInstance()
cam = py.InstantCamera(tlf.CreateFirstDevice())

cam.Open()
print(cam.DeviceModelName.Value)

temp_sens = cam.DeviceTemperatureSelector.Symbolics

for sens in temp_sens:
    cam.DeviceTemperatureSelector = sens
    temp = cam.DeviceTemperature.Value 
    print(f"{sens} temp is {temp}°C")

```

rohitrkr, 2022-05-24:
@thiesmoeller thnx buddy for your responses....
Partially Its working

thiesmoeller, 2022-05-25:
What part of "getting core board temp" is not working?

jvermaEt, 2024-09-26:
my system have multiple camera connected how should i get all camera temperature acA402024  usb

SMA2016a, 2024-09-29:
why do not open a new thread for new case?

Please refer document page 👍 
https://docs.baslerweb.com/device-temperature#python_2

if this code line fails to read the value, check with pylonviewer if your camera has got this feature