basler/pypylon #156: Cannot startgrabbing an image in windows 10, no errors, no image

StefanHope, 2019-09-24:
I can grabbing images in windows 7 by using a developed kit which is  Basler pylon x86 4.0.2.3531. And the type of camera is a GigE. The model is acA1600-20gm. 
When I worked on windows 10 x64 by using a Basler_pylon_5.0.12.11830, it can be conneted, but cannot grabbing imgaes, no any errors show in the messagebox.

something just like this:
[image]

Thanks!!!

MattsProjects, 2019-09-24:
Hi there,
This is a pretty common issue. With GigE cameras, this is usually either a firewall in the way, a camera packet size set too large to be supported by the nic/switch, or one of the camera's triggers has it's TriggerMode = On and it is simply waiting for a trigger (there could be up to three to check, AcqusitionStart,FrameStart,LineStart).
If you like, I have a diagnostics program that will check for these things automatically:
[](https://github.com/BaslerSupportUSA/CQT/releases/download/v19.09-01/CameraQuickTester_v19.09-01.zip)

StefanHope, 2019-09-25:
> Hi there,
> This is a pretty common issue. With GigE cameras, this is usually either a firewall in the way, a camera packet size set too large to be supported by the nic/switch, or one of the camera's [...]

Cool, thanks so much.

closed 2020-12-02 by thiesmoeller