basler/pypylon #158: How to setting camera Binning?

Fnska, 2019-10-07:
I cant figure out how can I get access to Binning Modes
```def main():
    try:
        camera = None
        
        camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    except genicam.GenericException as e:
        print(f'{e}')
        time.sleep(1)
        main()
    else:
        camera.Open()
        nodemap = camera.GetNodeMap()

        camera.BinningHorizontal = 4  # WORK
        camera.BinningVertical = 4  # WORK
        
        # camera.BinningVerticalMode = "BinningVerticalMode_Sum" # BUT THIS NOT WORKING
        nodemap.GetNode("BinningHorizontalMode") # RETURN: genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16599)
```
SetValue() also doesnt work

I always get `genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16599)` exception with two different methods call properties `BinningVerticalMode` or `BinningHorizontalMode`

thiesmoeller, 2019-10-07:
The mode for binning is not supported on all devices.
Check the manual on docs.baslerweb.com for your device

Fnska, 2019-10-10:
Thank you for your help. I inattentively read that If your camera supports only one binning mode per direction, the Binning HorizontalMode and BinningVerticalMode parameters are not available.

closed 2019-10-10 by Fnska